Peter-St/Android-UVC-Camera #1: Android AAR package?

laheller, 2019-03-29:
@Peter-St 

Once you get everything working, do you plan to create an Android AAR library with UVC camera isochronous transfer support?

BR,

Ladislav

Peter-St, 2019-03-31:
Hello Ladislav,

an AAR Library will be comming, but it will last some time to make this done. Some other things will have priority: to get a sample in the Play Store and so on. I will get to you back, when it's done.

So far,

Peter

laheller, 2019-04-03:
Hello @Peter-St 

Good to hear that!
Anyway, even when your app is done and published on Play Store, please let me know.

BTW my goal is to live broadcast the webcam stream (attached to Android device) to **youtube** or **facebook**. This is quite easy on Windows and Linux platforms using **ffmpeg** but I would like to do it on Android.

laheller, 2019-04-29:
@Peter-St 
Any news? Is there already app on Play Store?
Or you can send me app directly to test it.

Peter-St, 2019-04-30:
Hello Laheller,

I hope to get a beta of the app this week on play store.

Peter

Peter-St, 2019-05-11:
Today I released the app as a beta version in the play store. It should get online in a couple of time.

Greetings,

Peter

laheller, 2019-05-20:
Hello @Peter-St 

Just installed your app. It was able to detect attached USB/UVC camera and allowed to set me some parameters on my Huawei P20 Lite device but then the test failed and app crached.
Do you need to analyze any logcats or debug outputs?

Peter-St, 2019-05-20:
Hello,

send me the parametets which you set. It's normally, that the app doesn't work with the right settings. Only possibility to test all possibilities.

Petet

---- Ladislav Heller schrieb ----

Peter-St, 2019-05-20:
Hello,

send me the parametets which you set. It's normally, that the app doesn't work with wrong settings. On Android I made the Experience, that it depends more on the device, than on the camera, which settings to choose.

Best way is to try out differnt max packet sizes, diff. packets per request and diff. active urbs.
And if possible, try out other devices too...

Perhaps I should improve the way of the test run, because for now you only see if a frame is complete and not how this frame looks like.

And first thing is to get a valid testrun and than proceed to the streaming screen.

Peter

Peter-St, 2019-07-21:
Hello Laheller,

I have updated the app structure and fixed some crashes on late announced android devices and you can check out the app again, if you can perform a video stream and report it here.
But I'm still developing the app and doesn't know, if the app works with your Phone.

If you are able to test the app from Android Studio, then you could debug the app on your own. The Git Repo is still actual to the latest changes and works, if you import it on Linux PC.

Perhaps I will add a logfile for debugging the app, to make it easier to get it running on all devices in near future.

So far,

Peter

laheller, 2019-09-20:
> Hello Laheller,
>
> I have updated the app structure and fixed some crashes on late announced android devices and you can check out the app again, if you can perform a video stream and report it here. [...]

Hello

I will test the latest version, as soon as I get a new webcam, since the old is dead.

BR,

Ladislav

Peter-St, 2019-10-17:
And did you manage to get your Webcam working?

BW,

Peter