Peter-St/Android-UVC-Camera #2: Can't get camera to display feed

nldimartino, 2019-06-17:
I have connected a Logitech HD 720p C310 to the OTG cable and then into my Android Moto Z3 phone.  I can't get the camera feed to come up.  The app has crashed every time I've press "Transmission Start."  I have been switching the numbers for Packets per Request, Active URBs, and Max Packet Size and nothing has worked.  I would prefer not to do hundreds of combinations for this so do you have any tips on getting my camera to display the feed?

Peter-St, 2019-06-23:
Hello nldimartino,

I have updated the Testrun for the camera the last days, so please update you uvc_camera app and run 'Set Up The Camera' on main screen. 

Thirst set up the camera with uvc settings.
Then perform the controltransfer: 'Probe and Commit Controltransfer'
Please post a screenshot from your result.
Next perform under testrun an output of an single frame: 'Display One Frame'
Also make a screenshot from your output.
Next the Testrun: 'Display Frames for 5 secounds' and again a screenshot.

It would be easier to help, if I know your camera specifications, to compare your entered values with working values.

The MaxPacketsize depends on your entered Altsetting. For the first try choose the smallest MaxPacketsize, 1 ActiveUrb, and 1 PacketPerRequest.
Try to get the stream working.
Perhaps you can also post the output of:
' lsusb -v -d <vendor>:<product> ', if you have a linux pc, to know your camera specifications.Otherwise if the method 'Set up from uvc settings' works, please post a screenshot from the 'edit values' screen.

So far,

Peter

nldimartino, 2019-06-24:
[image]

Here are the specs for the Logitech webcam when I use lsusb -v -d in Linux.  I've been texting different numbers for Altsetting, MaxPacketsize, ActiveUrbm and PacketPerRequest from what you told me and I still cannot get the stream to pop up.  The app crashes every time. Can't check Logcat in Android Studio either to debug since the webcam is plugged in.

Peter-St, 2019-06-25:
Hello nldimartino,

I'm trying to trace the problem and think, it's perhaps you use a newer 64 bit device and this Device has doesn't load a Library the right way.

A solution for now would be, to test the camera from an older phone (with lower API) and look, if you get an output there. I will post here, if I can solve this issue.

So far,

Peter

nldimartino, 2019-06-25:
Are you saying the 64-bit phone I am using does not work with this app or that the device I am using is an too old to work with this app?  I would prefer not to have to change the phone I am using so I'm curious if this app works with newer 64-bit phones.

Peter-St, 2019-06-26:
Updated the app's libraries to support all architectures.

Please update the app and give it another try.

Peter

nldimartino, 2019-06-27:
Hi,

Thanks for fixing the app so that it does not crash all the time.  It runs a lot smoother now.  However, I am getting a message saying, ""Error: java.lang.Exception: Unable to open camera device connection."  I found this part of the code in SetUpTheUsbDevice.java on line 620.  It seems as though the camera I am using will not work correctly due to the kernel files not being patched.  Is that true?  Just want it cleared up.

Thank you,
Nick

Peter-St, 2019-06-27:
Hello Nick,

the moto z3 supports otg via usb. Try to connect a mouse or a keyboard via usb and look, if you can handle your phone with these input appliance.

To solve the issue please try following steps:

- unplug your camera and usb otg cable (hub) (unpower the camera too, if powered by a extern power source)
- reboot your phone to make sure, all functions of the phone works correct.
- eventually reinstall the app from PlayStore (perhaps not needed)
- start the app, connect the cam, load some values or set up the cam and do a testrun (normaly it should work then)

- Another (perhaps unlikely) reason for this exception could be the usb cable you are using. Eventually (if present) try out a usb otg hub, or another usb otg cable.

Please also post, if your are using Android Studio to run the app (you can git clone the actual app version from here), or if you use the play store app.

So far,

Peter

nldimartino, 2019-06-27:
Hi,

I forgot to mention that the app notices a camera is connected and it can list up the camera.  The error occurs when I press the "Set up UVC settings" button. 

I am running this on Android Studio.  I did all of the steps you mentioned.  I connected a keyboard to the phone and it worked.  It seems like the webcam might be the issue since the app sees it but UVC settings can't be applied.

Thank you,
Nick

Peter-St, 2019-06-27:
Ok, never mind.

Please copy the whole output of 'lsusb -v -d' in a '.txt' file a post it here.
You can also run the camera with your own settings with the 'Edit Save - your camera valus' button.
If the 'Set up with uvc settings' button doesn't work, it doesn't matters.
Write your own values from the linux command with the 'Edit / Save' button and you should be ready to go.

The 'Set up with uvc values' button doesn't support all cameras for now and because your camera has a microphone too, i think the output of the camera descriptor does not fit to the method for now. This will eventually been fixed later.

So far,
Peter

nldimartino, 2019-06-28:
Hi,
Here is the .txt file with the output from the lsusb -v -d on the webcam I am trying to use.
[WebcamInfo.txt](https://github.com/Peter-St/Android-UVC-Camera/files/3339477/WebcamInfo.txt)

I have reinstalled the app on my device and given it the right permissions.

Thank you,
Nick

Peter-St, 2019-06-28:
And still this Exception?

nldimartino, 2019-06-28:
Yes I still get the exception.  I am not sure if I am entering in the correct information for the UVC settings.  Any idea what they would be?

Peter-St, 2019-06-29:
ok, then let's continue with android studio.

Let's try to open the Usb Device with an Intet from Java (You have to gain the new permissions from your app to use usbHost Privilegs)
Please update your git repo and start android studio. Run the app in Debug, give all the permission to the app (Settings/App/UVC_Camera/Permissions) on your Phone and then proceed to the SetUpTheCamera Screen:

Now you have initialied a BroadcastReceiver, which detects, if you plug/unplug your camera. Normally you should have the permissions for your camera now (When plugging / unpluggin). If you don't have the permissions, click only on 'Find the camera' to request permissions.
The next step is opening the camera Device and running the Method 'Set Up with uvc Settings'
Please post the debug output from Android Studio.

Peter

Peter-St, 2019-07-04:
Is it working now?

nldimartino, 2019-07-09:
No, it is still not working.  I'm still just plugging in random combinations for the active URBs, Packets Per Request, and Max Packet Size and the camera transmission does not appear.

Peter-St, 2019-07-09:
Can you execute the 'Set Up With Uvc Settings' Button now without any errors?

nldimartino, 2019-07-10:
I cannot execute the 'Set Up With UVC Settings' button without getting the error.  I also cannot attach a .txt file of the exception because my phone can't be connected to the webcam and Android Studio at the same time to look at the log.

Peter-St, 2019-07-11:
Sure you can. Use Adb over wifi:
https://developer.android.com/studio/command-line/adb
from command line, or use a android app:
https://www.google.at/url?sa=t&rct=j&q=&esrc=s&source=web&cd=3&ved=2ahUKEwi81Yik8qvjAhUF_CoKHZo8Ce0QFjACegQIABAB&url=https%3A%2F%2Fplay.google.com%2Fstore%2Fapps%2Fdetails%3Fid%3Dcom.ttxapps.wifiadb%26hl%3Dde_AT&usg=AOvVaw3ZjMS8_1zc99ndMMBtpklW

to use adb over wifi, connect you phone in debug mode and activate the wifi.
On your pc type in following commands:

adb devices
adb tcpip 5555
adb connect <your phone's ip address>
(adb connect 192.168.0.1) for example.

Then you can unplug the usb cable and plug in your camera.

Peter

nldimartino, 2019-07-16:
Dear Peter,

I apologize for the delayed response.  The equipment I am using is not the easiest to work with and loading times using virtual machines take a long time.  Here is the logcat output for the camera when I press the 'Set Up The Camera Device' button.  I get a popup message that says 'permision denied for device.'  I have permissions allowed for the app so I am not sure how to clear up this permission.

[WiFiLogcat.txt](https://github.com/Peter-St/Android-UVC-Camera/files/3398337/WiFiLogcat.txt)

Sincerely,
Nick

Peter-St, 2019-07-16:
Hello Nic,

the problem is, that the permissions for the usb device were missing.
Do you get the checkbox:
' Allow the app UVC_Camera to access the Usb Device' and check the box for:
' Use by default for this Usb Device'
To receive the permissions, click on: 'Find the Camera' and update the files from the git repo again.

Try to execute all methods a few times and send me the log again to debug. Normally you should be able to grant the permissions for the camera.

So far,

Peter

nldimartino, 2019-07-17:
Dear Peter,

The only box that appears is the 'Allow Debugging' one.  I have granted permissions to the app but I am not able to access the camera permissions.  The 'Find The Camera' button does not appear to be working.  The Logcat output does not display anything when I press the button either.  However, I am able to List Up the Camera.  When I press the 'Set Up With UVC Settings' button, the only text that appears is 'No Permissions were granted to the Camera Device.

I am not sure why I am not able to grant permissions from my camera since the phone does not offer a way to grant them from what I can see.

Thank you,
Nick

Peter-St, 2019-07-17:
Hello Nic,

please try out any other camera apps (CameraFi,...) from Play Store and report, if you get there the popup window, to grant the usb permissions and can check to remember them.

So far,

Peter

nldimartino, 2019-07-18:
Dear Peter,

I tried out CameraFi.  I am able to get permissions for everything to work.  The app will display my webcam feed but the app will crash soon after.  Probably a problem with that app.  However, the camera feed is able to be displayed.  I then went to your app and it says the permissions were denied.  

Sincerely,
Nick

Peter-St, 2019-07-19:
Hello Nic,

I think i have solved to Problems with the permissions. Now I'm using an intent to find the camera and gain the permissions.
First update the app, next connect the camera and agree to the apps popup, when it asks for starting the service (Permissions should then be granted). Please update and give feed back over the result.

Peter

nldimartino, 2019-07-22:
Dear Peter,

Did you rename the app to humer.uvc_camera?  I am not able to open the app now that I have updated it.

Sincerely,
Nick

Peter-St, 2019-07-22:
Hello Nic,

1st.) Deinstall the app, install it from Play Store and report, if you can start the app.

2nd) Try to install it via Android Studio and attach the output of the log.

Pls report when using the play store version, if you get a pop up window, to start the app, on your device, when you plug in the camera. (you should have permissions then to the usb device)
And if you can use the play store version.

My last two commits didn't change the app name (I never changed the app name up to now), so there must be an other reason for the crash....

In the Play store stack trace there is an Exception (Resources$NotFoundException) only on Android 9 Pie. I have added a new Resource file, which should detect your Camera in my last app commit. I started the app on Android 6 and it worked flawlessly.

So far,

Peter

nldimartino, 2019-07-23:
Dear Peter,

I uninstalled the app and deleted the app files from my computer.  I downloaded the app from the Play Store and also on Android Studio without the Virtual Machine.  Both of these paths resulted in a message saying that the device permissions were denied.

I then downloaded the app through the Virtual Machine on Android Studio and I can't open the app when the phone is directly connected to the computer and connected over wi-fi.

It appears that the webcam device I am using will not allow me to access its permissions.  These issues do not seem to be a problem with your software but a problem with the design of the device.

If you have any other thoughts that would be great.

Thank you,
Nick

nldimartino, 2019-07-23:
Dear Peter,

What types of USB cameras have you used with your app that were successful?  I would like tot est your app with a camera that you have personally tested.

Sincerely,
Nick

Peter-St, 2019-07-24:
Hello Nic,

in general I think, it could be a reason of the high android version too and not only the camera device.

If other apps (CameraFi) works with you device, this app should work too and perhaps there were added some usb code in Android Pie, which makes it harder to connect to the camera.

You can try your camera with an older android device (if present) and report your results here. My app works on devices down to API 15 (Ice cream sandwich).

Some questions to the latest update:

1) Are you getting a popup window now, to start the app, when you connect your camera.
2) Also attach the latest log from Android Studio, when debugging the device: (have you added the new file in your res folder in Android Studio? --> /app/src/main/res/xml/device_filter.xml  (path of the new file).
3) Let's try to get Android Studio working again, because it's easier to debug with it.

So far,

Peter

Peter-St, 2019-07-24:
One way to obtain the permissions could also be over the camera 2 api, which is also able to communicate with an external usb camera. I will test it out on WE.

Peter

Peter-St, 2019-07-29:
Update:

I was able to test the app with Android 8 by now and found a bug in the Storage Permissions. You can update the app again and test it.

And another problem needs to be solved:
With the newer Kernel Sources on Android this app needs some updates in processing the isochronous stream. I'm receiving a lot of errors on Android 8, which I don't get on lower Devs. I need to update the program structure to fit to new kernels.
For now I'm not able to use this app on Android 8 Device's. Hope to get it working soon.

So far,

Peter

Peter-St, 2019-08-02:
64 bit support is now online. You can give the app another try.

hdupont1, 2019-12-14:
Hi, I try to set up the app for my uvc camera Alcor micro 5102 USB 2.0, with no success. Message is setup not correct. Do you have any idea how to resolve the problem. Thank you in advance for your help.

Peter-St, 2019-12-14:
Hello Nic,

run the button "Set Up with Uvc Values" and then all the values should be correctly set.

Save your values under a name and try to run the "Testrun" Button.

Please post your output and values.

So far,

Peter

hdupont1, 2019-12-14:
Hello, if I run the button "set up with uvc values", it is not automatic, I have to select the values.

Peter-St, 2019-12-15:
Hello Nic,

then lets make the first example for your camera:
Run the SetUpWithUvcSettings Button with following inputs:
MaxPacketSize: select something between 1000 and 3072.
PacketsPerRequest:  Select for the Start 2
ActiveUrbs: Select also 2
Select the Frame Format: YUYV
Next Select the Resolution with: 1280x720
And at last the frameInterval with: 666666, or 1000000.

Please report the testrun or other values you use for your camera.

So far,

Peter

hdupont1, 2019-12-20:
dear Peter, i thank you for your reply. now it works with your recommendation, except i replace yuyv by mjpeg. but the images are not fluid. 1 image every 1.5 sec. which parameter i should modify in order to improve the fluidity. i do not care about the quality of image. for your information, i have this camera:
  https://m.tomtop.com/p-k6569.html

Peter-St, 2019-12-20:
Hello Nic,

you sucessfully received the camera stream for your settings. (I don't know the exact settings you choose, perhaps you add a screenshot of the main screen.)
 It looks like your camera supports more than one Frameformats (YUY2 and MJpeg). If it works with Mjpeg you can stay at Mjpeg. The two values you have to change now are ActiveUrbs and PacketsPerRequest. Select for Example 16 for both of them and perform a testrun. (Perhaps you add also a screenshot of the output of the testrun). Look at the output of 16, 16 at PacketsPerRequest and ActiveUrbs. Next try out 4 ActiveUrbs and 16 PacketsPerRequest and compare the output in the Testrunlist with the previous Output (you can also start the VideoStreaming and look there, if it gets more fluid).
Next try 16 ActiveUrbs and 4 PacketsPerRequest. Also compare the output. 

When you compare the output you have to take a note on the frame length. Post the number you receive from the testrun. When the frames get longer, you get better Pictures with less errors from your camera. All the frames with errors are canceled automatically by the app.

Everything clear?

Peter

closed 2020-12-28 by Peter-St