Peter-St/Android-UVC-Camera #3: Add a license

dllud, 2019-10-20:
Thanks a lot for this app, much useful.

Could you please add an open-source software license to your code? Check out: https://choosealicense.com/

Without a license, as it is, all your code is under copyright, which means that [no one can copy, distribute or modify it](https://choosealicense.com/no-permission/). Is that what you intend?

Peter-St, 2019-10-20:
Hello dllud,

I will add a licence to my repo, but before I have to get the permissions of other repos I'm using in my code, to get a valid license.

So far,

Peter

Peter-St, 2019-11-10:
Hello dllud,

I have added the License to this repo. You can now use this repo to distribute or modify it.
I'm going to close this Issue for now.

So far,

Peter

closed 2019-11-10 by Peter-St