Peter-St/Android-UVC-Camera #4: Camera Autofocus

costeacostin, 2019-11-06:
Hello,

Great job for this project, it was really useful for me after some magic params combinations. 
I use an UVC camera that is trying to autofocus. 
Do you think that it can be any chance to control the autofocus? at least to enable/disable it

Cheers,

Peter-St, 2019-11-07:
Hello costeacostin,

thanks for using my app. I can handle the thing with the camera auto focus, but this will last some time. I'll get back to you, when it's done.

Peter

costeacostin, 2019-11-08:
Thank you a lot. It will be really useful. If I will succeed to do this on my own I will also come back to you. Also I guess that we can use multiple cameras with these configurations (just a question)

Peter-St, 2019-11-10:
Hello costeacostin,

I tried to add the autofocus for the camera, but I was not able to test the function. Please update the app from the repo, or reinstall it from Play Store and report, if the Auto Focus works.

So Far,

Peter

Peter-St, 2019-11-12:
About multible cameras with these configuration, I will test it the next days ...
And if it works, I think I will update my App with this feature, but it will sure last some time.
Peter

costeacostin, 2019-12-15:
I tried and it works, thank you a lot mate!

closed 2019-12-15 by costeacostin

Vanntreudy, 2020-11-13:
Hi! I am very new to android UVC Camera. I tried to set function getAutoFocus() auto when UVC Camera start, but I can't do. I don't where to start with. Should I config in Lib UVC Camera? Or where? Please Help me