Peter-St/Android-UVC-Camera #5: WebRtc USB camera support

SilvanaOlteanu, 2019-12-15:
Hi Peter, 

Thank you so much for posting this project, it is really helpful. 
This is not really an issue, I was just wandering if you tried sending the frames captured by the camera to another device using webrtc. I've tried for some time to implement this but with no luck and I'm really curious how I can do this. 
If you have any info regarding this, it will be very much appreciated.

Cheers,

Peter-St, 2019-12-15:
Hello Silvana,

up to now I don't have any experience on WebRTC, but a similar problem is discussed here:
https://github.com/saki4510t/UVCCamera/issues/9

Perhaps I will test it in near future and will come back to you.
If you solve this issue plz report.

Thx,

Peter

SilvanaOlteanu, 2019-12-18:
Hi, 

Yes, i tried that but ran into some deprecated functions issues. Will continue to investigate this  and if i find something, i will let you know. 

If you test it, please do tell. 

Thanks, 
Silvana

MelvinFMQ, 2020-02-15:
I'm also facing this issue, the function is no longer used in webrtc

Peter-St, 2020-04-12:
Hello,

WebRTC is now supported with YUV Format Cameras.

Peter-St, 2020-04-13:
I also added a support for Mjpeg frames for now.
WebRTC only accepts frames in NV21 Format, so I had to include some image manipulation too.
This increased the app size to double size. Perhaps this manipulation can be solved better (without OpenCV).

The version of WebRTC I imported to the project is 1.0.22672, which is supporting the old function: onByteBufferFrameCaptured, to send the frames to WebRTC.

Peter-St, 2020-12-28:
The WebRtc Support is granted for now. The importation of the lib is quite old, but it's working for now.

closed 2020-12-28 by Peter-St

mohdqasim, 2021-01-11:
> The WebRtc Support is granted for now. The importation of the lib is quite old, but it's working for now.

@Peter-St if possible can you right article on this to get more clarity

Peter-St, 2021-01-11:
Hello mohdqasim,

there are newer approaches to work with webrtc and usb cameras. My approach is working, but the webrtc library I'm using is deprecated since some years. I haven't had time to update it to a newer versions, but I will do this in near future.

There also have to be some improvements on the image conversations in the native code of my app, because they are working, but have some malfunctions too.

So far,

Peter