Peter-St/Android-UVC-Camera #6: "Can't get Camera Permissions"

satmandu, 2020-01-08:
Using the current application from the play store, can not setup the camera. I have my USB OTG camera working with other apps. 

The only permissions for the app which can be granted appear to be for storage.

In contrast other apps which show the camera feed also have a camera permission...

I'm on Android 10 on a Pixel XL.

Peter-St, 2020-01-08:
Hello Satmandu,

for your question I give you the answer from Saki4510t : 

> Android 9 and more need CAMERA permission to get USB permission for UVC devices.

> And unfortunately Android 10 has critical issue on granting USB permission now, pls check Issue Tracker:
> https://issuetracker.google.com/issues/145082934

This means, that there are security issues on Android 10 and there is no way to grant the Usb Camera Permissions for Android 10 Devices (specially for Google Pixel Devs).

For now one Solution for you would be, to use my App complied with a lower Sdk Version than 28. The App on Play Store needs to have at least Sdk Version 28, otherwise it can't be uploaded.
Over the following link you can download and use my app still with Android 10:

https://drive.google.com/open?id=1EJl3KgfS0ksWr14dgRqv4yDkF2Bo1OBm

The only difference from the actual app is, that this app is compiled with Sdk 27 (Android Oreo) and here the latest security updates from Pie and Q are not included.

Please also report which current Apps from Play Store you also use and where you can grant the permissions.

So far,

Peter

satmandu, 2020-01-08:
Thanks. That does avoid the error.

This app appears to load correctly, so maybe they worked around it?

https://play.google.com/store/apps/details?id=com.shenyaocn.android.usbcamerapro

Peter-St, 2020-01-09:
Hello Satmandu,

the app you showed me was from the guy, who made the Usb Permissions Issue on Google Issue Tracker.

Maybe I'll get some info, about handling it with higher Sdk versions.

So far,

Peter

Peter-St, 2020-12-28:
The camera permission issue should be solved with the latest update (available on Play Store and over this repo

closed 2020-12-28 by Peter-St

AshikLanjewar, 2024-07-02:
Hi All,
   If anyone still facing the issue regarding request permission issue on Android sdk 28 and greater than that then follow below steps

1.  Add Extra Permission for device camera in manifest.
    Use below code:
    `<uses-permission android:name="android.permission.CAMERA" />`
`<uses-feature`
 `        android:name="android.hardware.camera"`
 `        android:required="true" />`

2. Before start USB or UVC camera, We have to check permission on runtime. If permission not granted then ask for the     
    permission using below code

   `private void checkCameraPermission() {`
   `		// Add permission for camera and let user grant the permission`
   `		if (ActivityCompat.checkSelfPermission(this, Manifest.permission.CAMERA) !=` 
   `PackageManager.PERMISSION_GRANTED`
   `			&& ActivityCompat.checkSelfPermission(this, Manifest.permission.WRITE_EXTERNAL_STORAGE) !=` 
   `PackageManager.PERMISSION_GRANTED) {`
   `			ActivityCompat.requestPermissions(this,`
   `				new String[]{Manifest.permission.CAMERA, Manifest.permission.WRITE_EXTERNAL_STORAGE},`
   `				REQUEST_CAMERA_PERMISSION`
   `			);`
   `		}`
   `	}` 
	
3.  After responding on permission popup, If any stuff you have to execute on permission grant or denied then use below 

    code snippet
   `@Override`
   `public void onRequestPermissionsResult(int requestCode, @NonNull String[] permissions, @NonNull int[]`
   `                                                                 grantResults) {`
   `          super.onRequestPermissionsResult(requestCode, permissions, grantResults);`
   `          if (requestCode == REQUEST_CAMERA_PERMISSION) {`
   `                  if (grantResults[0] == PackageManager.PERMISSION_DENIED) {`
   `                                //close the app`
   `	                        Toast.makeText(this,"Sorry!!!, you won't be able to use this app smoothly without this 
   permission.",Toast.LENGTH_LONG).show();`
   `                  }`
   `           }`
   `}`

4. Once user granted the above **Camera** permission, then request for **USB_PERMISSION** , it will work like Android SDK below 28.

   Please check android official link in details: 
[Developer Website link](https://developer.android.com/reference/android/hardware/usb/UsbManager.html#requestPermission(android.hardware.usb.UsbDevice,%20android.app.PendingIntent))