Peter-St/Android-UVC-Camera #8: Automatic detection generate error

costeacostin, 2020-03-30:
Hello,

I tried to set up the UVC parameters with the automatic params detection (which I found pretty interesting btw) and it fails.
I tried to debug it and so on but I didn't found anything relevant.

E/AndroidRuntime: FATAL EXCEPTION: main
    Process: humer.uvc_camera, PID: 7391
    java.lang.NullPointerException: Attempt to read from field 'int humer.uvc_camera.UVC_Descriptor.UVC_Descriptor$FormatIndex.numberOfFrameDescriptors' on a null object reference
        at humer.uvc_camera.SaveToFile.selectFrameIndex(SaveToFile.java:1022)
        at humer.uvc_camera.SaveToFile.selectFormatIndex(SaveToFile.java:966)
        at humer.uvc_camera.SaveToFile.selectUrbs(SaveToFile.java:911)
        at humer.uvc_camera.SaveToFile.selectPackets(SaveToFile.java:874)
        at humer.uvc_camera.SaveToFile.selectMaxPacketSize(SaveToFile.java:836)
        at humer.uvc_camera.SaveToFile.setUpWithUvcValues(SaveToFile.java:758)
        at humer.uvc_camera.SetUpTheUsbDevice$23.onDismiss(SetUpTheUsbDevice.java:958)
        at android.app.Dialog$ListenersHandler.handleMessage(Dialog.java:1408)
        at android.os.Handler.dispatchMessage(Handler.java:106)
        at android.os.Looper.loop(Looper.java:166)
        at android.app.ActivityThread.main(ActivityThread.java:6861)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:450)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:936)
I/Process: Sending signal. PID: 7391 SIG: 9

BR,
Costin

Peter-St, 2020-03-30:
Hello costeacostin,

thx for debugging the app.
This failure is because your camera supports more than one format indexes, but this case isn't included so far in my app.

I will solve it soon.

Peter-St, 2020-03-30:
ok, plz update your app from github and try it out again ...

Peter-St, 2020-03-30:
ok, plz update your app from github and try it out again ...

Note:

The Automatic Detections is in beta Development for now. A lot of cases of setting up the values for different cameras on different devices are not included for now.

Peter-St, 2020-12-28:
The latest Driver of this app works better with every camera. The Auto Detection will remain in Beta I think, because it's heavy complicated to get every camera with every phone runnung, without any malfunctions.

closed 2020-12-28 by Peter-St