Peter-St/Android-UVC-Camera #9: can't to get uvc permission in galaxy(android 10 series)

Jade-source, 2020-04-02:
I tried connect webcam on galaxy s9, s20(android 10). 
As a result, it was unable to connect due to the denied authority.

So i tested on pixel(android 10) smartphone. Because to check whether it is an OS problem or a manufacturer problem.
As a result, pixel(android 10) is able to get permission for UVC(USB Video Class : 0x0E).

Peter-St, 2020-04-02:
Hello Jade-Source,

it's a bug from google.
Please look at:
https://github.com/saki4510t/UVCPermissionTest/issues/1

So far,

Peter

Jade-source, 2020-04-03:
I confirmed saki4510t's issue.

Thank you Peter-St

closed 2020-04-03 by Jade-source

Jade-source, 2020-07-15:
Hi, Peter

some good news.

UVC permission issue be able to solved on Samsung galaxy(andorid 10 series).
I made a request to the Samsung development team.

Samsung replied that it would solve the issue.
"The estimated schedule for the fix in SAMSUNG devices would be early AUGUST 2020 or early SEPTEMBER 2020."

Peter-St, 2020-07-15:
Cool 😎