Peter-St/Android-UVC-Camera #10: Can't skip the data loss frame in the IsochronousStream class

Jade-source, 2020-04-03:
Hi, Peter-St

I found that in the Galaxy(android 9 series) accepts packets with data loss without skipping them.

As far as I've checked, IsochronousStream Class can't skip the problematic frame.
The reason seems to be the lack of defensive code to detect frames.

In IsochronousStream class, the current header length, flag, and data length are only detected,
So frame detection is improved by adding FID and EOF, but skip is not perfect.

[image]

Peter-St, 2020-04-03:
You are right.

I experimented some time ago on this behavior, but wasn't able to solve it.
Any help would be welcome to solve this issue.

Peter-St, 2020-12-28:
The latest Approach over the LibUsb Driver should solve this issue. Update the app to the latest version and set up the Camera with the LibUsb driver.

closed 2020-12-28 by Peter-St