Peter-St/Android-UVC-Camera #11: setup for Huawei P8 with AN97

ErnstTom, 2020-04-29:
Dear Peter Stroiber
now i have some time to try the UVC app.
but a lot of things are new to me, unknown.
How to set up a Huawei P8 (non-lite) mobile phone with AN97 endoscope.
i have tried many apps but all of them only work with the high-end and samsung device only.
and I think the huawei p8 isn't weak either

Peter-St, 2020-04-29:
Hello,

up to now I wasn't able to test these Kirin Processors with my app, so I can only give you some advices.
I think this app could work with your mobile phone, but I don't know how good or bad the usb-source of your phone is.

Basically you start to detect the camera. You take a look at the Interface's and Endpoints of your endoskop, to get sure it's a uvc device. And at last you try to get some data of your camera.
For getting some data you use the manual uvc set up method.
You can choose the lowest values for the resolution.
If you start with 1 PacketsPerRequest and 1 ActiveUrbs you need to find the right Maxpacketsize (/Altsetting) for your devices.
So you try out all the Maxpacketsize's of your camera and start a testrun.
The testrun where you get the most bytes back from your camera, is the right one.
You keep this Maxpacketsize and try then to raise the PacketsPerRequest and ActiveUrbs. Perhaps only to an amount of 4, perhaps higher. When you notice, the bytes gets smaller, or the stream fails, you have to lower these values again.

Good luck.

ErnstTom, 2020-05-03:
> &mdash;

Dear Peter Stoiber,thank You for quickly answer The device (endoscope) is detect and the usb is permissions.
On screen of Huawei P8 is this attached screens. (possible choice offered)But by Testing run com an alarm There are places where you do not offer a value that you have suggested.
You would take a picture of YUYU.
But I hope I don&#39;t bother you with my questions. Ernst Tom

Peter-St, 2020-05-03:
Please post your values or adjust a screenshot from your values.

ErnstTom, 2020-05-03:
this 3 attached pictures (1,3,4) are printscreens of my huaweithe pictures 4 is printscreen of my device too, but for some explanation, but I don&#39;t understand

Peter-St, 2020-05-03:
Please post the pictures, or give a link to the pictures...

ErnstTom, 2020-05-03:
Oh, I&#39;m sorry.
I attached 4 jpeg printscreen to both of my last letters. I didn&#39;t know he didn&#39;t go through. I haven&#39;t had a relationship with github yet.
What is the best, easiest solution. And then I&#39;ll do it tomorrow. Just keep calm about me. Not so urgent, it would be good if I could use the endoscope not only connected to a PC but also with a mobile phone.
And I&rsquo;m still trying it - I didn&rsquo;t have time for it over the weekend.E. T.

Peter-St, 2020-05-03:
Load the screenshots up to google drive and post the link.
Or you click on the pictue icon direct in the github dialog menu, select an item and post it.

ErnstTom, 2020-05-04:
the attached:
[image]
[image]
[image]
[image]

Peter-St, 2020-05-04:
Hello Ernst,

the values from the 1st Picture should fit.

The 2nd Picture does not matter up to now.

In the 3rd and 4th Picture your Value for PacketsPerRequest and ActiveUrbs are much too high.

Start with 1 and 1 for those two Values.

Normally you do not have to edit the values in the edit/save menu.

You can try out then the -Controltransfer-Testrun- Button and attach the output again with a screenshot.

ErnstTom, 2020-05-04:
Dear Peter St.
[image]
[image]
[image]
[image]
[image]
[image]

I tried a lot of settings all the way through - not all the variations.
and the flash drive (10) and camera (11), which run on a pc, are recognized by OTG usb.
but an error comes (14,15-failed to start the camera stream).
Maybe you really can't use the device?
You have a token that is a non-Samsung mobile,
maybe even another type of Huaweien running the app?

Peter-St, 2020-05-05:
Please post the output of the first menu (Video Probe-Commit Controls) in the -Controltransfer-Testrun- menu with your selected values.
So far,

Peter

ErnstTom, 2020-05-05:
Dear Peter St.
I send the printscreens
[image]
[image]

maybe it's only OTG capable but not UVC support?
I just asked the Hungarian huawei if they think he knows the device.

Peter-St, 2020-05-05:
If you have the possibility to debug your phone over android studio with a PC / Laptop, we get exacter outputs, whats failing.

The thing what fails is the control transfer sended to the endoscop. Perhaps a weak OTG adapter (I also had one) If you have another otg hub, try it.

Eventually, if you have a linux machine on you Pc, you can post the output of:
lsusb -v -d xxxx:xxxx
First xxxx is vendor id / Second xxx is the product id.

Peter-St, 2020-05-05:
Hui, ok,

We got 1 working controltransfer.

The value 614400 was sended from your endoscop device and fits so far...

Peter-St, 2020-05-05:
One thing is strange, but should not matter up to now --> Your Hint value changed from 1 to 0 after the first control transfer

Peter-St, 2020-05-05:
I will give you a link with a debug apk, which you can install on your Phone and then plz run the same method with the same values ...

ErnstTom, 2020-05-05:
There are only win98 and 7. (PC, LT).
I don’t understand exactly what to do - I don’t understand that much on mobile android. Or should I read something connected to pc / lt?
Original (added cable, connector - but I tried other USB A / USB C converters as well). In the case of flash drive, ok.
AN97 endoscope available in the Far East.
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

Peter-St, 2020-05-05:
Plz wait, Till I get the apk finished ...

Your endoscop camera is an UVC device.
We have to send 4 Controltransfers to your Endoskop Camera to Set it up with your Selected Values. --> The process is automatically handled by the app, but fails in your case.

Peter-St, 2020-05-05:
Download this apk and run it with the same values.

https://drive.google.com/open?id=1lJ9hJyqAg-EqPGJUPZpdk_bN7gEPS2Jk

Peter-St, 2020-05-05:
Again attach the output of the first menu (Video Probe-Commit Controls) in the -Controltransfer-Testrun- menu with your selected values.

ErnstTom, 2020-05-05:
Thank You
I make that, meine ich, morgen kann ich darüber eine Rückmeldungen senden

ErnstTom, 2020-05-06:
answer from Huawei:
"In our experience, the Endoscope cannot be connected to the P8.
For compatibility with other external devices, I suggest that you contact your manufacturer or distributor. "

ErnstTom, 2020-05-06:
I deleted the old one last night and installed the new one - but unfortunately the result is the same.

[image]
[image]
[image]
[image]
[image]

Peter-St, 2020-05-06:
Hmm,

At this Picture:
[image]
you got a valid return from your endoscop.
Try to receive the same output again.
Perhaps uninstall and reinstall  from Play Store, to test the old version too.

How you got the working Output?

ErnstTom, 2020-05-07:
ALL RIGHT.
although the first page of the image you mentioned (05.05.17: 07) also showed an error while initializing the camera.
(the image comes with the original (playstore) version)
[image]
[image]
[image]
[image]

ErnstTom, 2020-05-07:
Dear Peter
but surely there is a fault in one of my devices.
I've already kidnapped you a lot of your time.
although if it came together, many huawei owners would surely be happy.

Peter-St, 2020-05-07:
Hmm,

if you could also spend some time, you could install android studio on your PC and try to debug this issue by your self.
I would update a Win Version of my github repo, because this on is for Limux, don't know if you could load it on Win.

The other way would be to test apk's uploaded by me. You have to give me then each time feedback. --> it's also not clear to me why the currently linked apk don't have the same return, like the one on Play Store.

Choice is to you.

ErnstTom, 2020-05-08:
Sorry, but I can't identify based on the text what and where to download for pc testing.

Peter-St, 2020-05-08:
Ok, then we will try to debug over sample apps, which I will upload.

Explanation:
The source code in this repo was written with Android-Studio. This is the most common tool for writing apps for Android Platforms. Other Programs for App Building for example are Eclipse or Netbeans or...
If you have no experience in Android-Studio for now, it will be much too hard for you to set it up and next try to debug an app on your own.

Peter-St, 2020-05-10:
Please test these 3 apks and post only the output, if you receive a sucessful return from the endoscop.
Camera 0:
https://drive.google.com/open?id=1-Ss0H9swlJcq6xOmeigb3W_NdN3XaUlu
Camera 1:
https://drive.google.com/open?id=13KsZKfuBeOH0h2O4wyzI_2DtcU0269x7
Camera 2:
https://drive.google.com/open?id=17UIye2qzvH0lvzjUHwkWVUYyVyA8jxFA

So far,

Peter

ErnstTom, 2020-05-10:
Dear Peter
I made 2-2 versions for each app. I hope my identities are understandable.
Actually and unfortunately I don’t know which variable would be “vital” and which is just a refinement.
I downloaded the Android studio (almost 1 Gbyte) but haven't installed it yet.
I'm afraid the device will forbid the use of such a camera on the USB port.

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

Peter-St, 2020-05-10:
hui, don't know how to solve this issue for now.
Run the last apk on your device and post the output:
https://drive.google.com/open?id=1PzPjNN_edxH8Cab1Zq7SERRPNA1t81Qb

You can try this:
Connect some other usb uvc camera to your Phone. Run the app and see if you can get there an output. I think the error is because of the endoscope camera and perhaps not because of the phone.

Post only the output with the controltrasfers, because here we need to get 4 successful returns.
The 4th control transfer returns:
-1
That means here we got the error. And this we need to solve.

For Android Studio:
Try to create an app for your self. Try to find some Tutorial for building an android app on Win and try find find another one for setting up android studio on Windows.
For debugging an usb app on Android you need to use the adb bridge over wifi to your phone, because you need the usb connection for your usb camera. There also exists some tutorials.

So far,
Peter

ErnstTom, 2020-05-11:
Dear Peter
 always set the same variable value - maybe I should change one of the others?
the 3-xx is another usb camera that also works on pc and is in a similar category.
Respectively: my other phone is not otg capable.
I can’t really ask my friends who might have it to put software on it and try it out.
BUT it is also possible that I will invest in a samsung and newer device in the near future.
This camera issue isn’t vital, it would only come in handy during my work (it would have come and I bought it) if I could use it in a place where I can’t set up a laptop right away - and I don’t even have one.

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

Peter-St, 2020-05-11:
Hmm,
the output of the secound camera looks also not valid.
It could be the otg adapter. Eventually try to use an usb otg hub.
I had an otg adapter which also not worked -->
https://www.amazon.de/gp/aw/d/B072L6LBJF/ref=yo_ii_img?ie=UTF8&psc=1

This hub for example worked with some of my old phones, while other one doesn't -->
https://www.amazon.de/gp/aw/d/B00L32UUJK?psc=1&ref=ppx_pop_mob_b_asin_image

Another Phone device would also be interesting, but I don't think it's the Samsungs fault.

So far,

Peter

Peter-St, 2020-05-11:
This output looks very stranges
[image]

The endpoints are the same (2 times 5) and the max packet size also --> doesn't make sense.
The value of each endpoint also is different.

I have also written the same program which you use on your android phone, for a PC too.
I will also try to send you a working program for your pc  where you can read out the values and the compare them. --> could last some time...

ErnstTom, 2020-05-12:
Recommended is not currently available.
Even when ordered from elsewhere, delivery is slow for the time being.
We may have to give up so much investment here after work now.
Until I find another device (a good acquaintance) where I can try things out, I might have to pause the attempts.?
Where and how could I thank you for your efforts and help in the meantime? Which I don't think many people would have done. 
Because this app is adjustable against all the others, I was hoping it might come together. 
I got to CNC, but unfortunately not for that.
But in doing so, I got to know such a community (github) and helpfulness.

[image]
[image]
[image]

Peter-St, 2020-05-12:
Please post a picture of your secound camera too.

In the case of the endoscope, we could try to start the stream directly, without calling the controltransfers (not logical, but we could try). Therefore I have to rewrite some source code (could last some time).

Then to your Win PC:
If you have time, you could try to read out usb descriptors on your win pc:
Follow this guid in thid tutorial:

Your need to install libusb on your win PC:
https://github.com/libusb/libusb/blob/master/INSTALL_WIN.txt

To install libusb on wimdows try out the following:
https://github.com/libusb/libusb/blob/master/INSTALL_WIN.txt
under the section:
> Installing and building libusb via vcpkg
> ****************************************

Try to run these methods, if you can complie it sucessfuly:
listdevs.exe
xusb.exe -i xxxx:xxxx

So far,

Peter

ErnstTom, 2020-05-13:
Dear Peter
This is a pen camera that I currently use as a microsd card reader or can even be a usb camera 
for skype or viber, for example. Windows did not require its own disk to install.
The surveillance camera is not recognized.
I don't have microsoft visuals.
I downloaded the vcpkg master. But this science is too much for me. I can't follow events anymore.
I also do correspondence using google translator. But I admire your perseverance. 
Although, if it worked, many Huawei and Honor owners would be happy - presumably based on app posts.

[image]
[image]
[image]

ErnstTom, 2020-05-15:
oh yeah, i forgot the avis feedback to the previous one that i started vcpkz.bat but nothing spectacular happened.
Should the android studio have been launched as well?

Peter-St, 2020-05-15:
For getting the Descriptors of your Usb devices you can use this program in windows:
https://www.thesycon.de/eng/usb_descriptordumper.shtml

> Download tdd.exe

Please copy the hole descriptors in a txt file and adjust them.
You can adjust them for both cameras.

ErnstTom, 2020-05-16:
both device work (gives a picture) - viewplaycap app for win.
it is interesting that the pen camera now only works with the card in it, 
otherwise it does not give an image, it is not detected by the program.
with card, the tddexe program gives an error, but I see the image on the interface.
[DescriptorDump_USB2.0_endoscope.txt](https://github.com/Peter-St/Android-UVC-Camera/files/4638097/DescriptorDump_USB2.0_endoscope.txt)
[DescriptorDump_USB2.0_pencam.txt](https://github.com/Peter-St/Android-UVC-Camera/files/4638098/DescriptorDump_USB2.0_pencam.txt)

ErnstTom, 2020-05-16:
could the mobile (Huawei P8) be tested instead of the camera somehow?
because there may not be some permissions there.
and then our effort is unnecessary.

Peter-St, 2020-05-16:
Please test the following app with both usb cameras:
https://drive.google.com/open?id=1I2RRTb9wcW1YEsBLbCBt2Cq5nY4TeYxm
New in this version is a new method under the -Controltransfer-Testrun- Button.
First Set up your Uvc Values with the Manual Method --> 1 Packet and 1 ActiveUrb // Then click on "-Controltransfer-Testrun-  // Test the Stream" to start the stream anyway.
You get an log file in your /UVC_Camera/log directory on your mobile phone.
Each time you click the button, the log file will be overwritten. Adjust this log file for each camera plz.

Peter-St, 2020-05-16:
Run this app only with the endoscop:
https://drive.google.com/open?id=17nOyAP5yjDeyMmx-ZmEPAEytiVyTCxQU
Set up the endoscop manual and click "Test the Stream"
Adjust the log file.

Peter-St, 2020-05-16:
> could the mobile (Huawei P8) be tested instead of the camera somehow?
> because there may not be some permissions there.
> and then our effort is unnecessary.

Hmm, I think it's in both cases a fault of the camera.
endosope camera:
Here exists a difference between android and win platform:
In the description of the endoscope stands, that there are different resolutions possible between win and android. So it could be, that this endoscope is specialy set up by an android device.

pen cam:
here we have a lot more features than a normal cam has. The pen cam has an audio input and a sd card input. For now I was only able to test cameras with only video output.
You can test test the pen cam with sdCard inside and Outside also with the latest apk attached and post then the log file.

So far,

Peter

ErnstTom, 2020-05-16:
endoscope details now:
the other doesn’t want to work now - my friend used to say Bill Gates.
with and without a card, the device now work on the pc.

with card the android recognizes it and handles it as a flash drive.
with or without card on android recognizes it 
but uvc manually stops it with an error and 
also resets to null a previous data state in test the stream.
that is, he is now dead. therefore, testlog also has only one.
I try everything twice (three times).
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[endoscopelog.txt](https://github.com/Peter-St/Android-UVC-Camera/files/4638426/endoscopelog.txt)

ErnstTom, 2020-05-16:
dear Peter, wait,
You're faster than me.
i did the last testing on may 16 at 13.10
i also do one with 14.24.
in the meantime, relax a little

ErnstTom, 2020-05-16:
this data is with app5 14.24.
the results look the same for me.
the pen has the same error and reset as with app4.
[image]
[image]
[image]
[endoscapp5log.txt](https://github.com/Peter-St/Android-UVC-Camera/files/4638493/endoscapp5log.txt)

Peter-St, 2020-05-16:
Please try to run the Manual Method with this app on your Pen device:
https://drive.google.com/open?id=1C09tx3wuxH7oSmxBo9Ji-Bwxp-mViUy1
Attach the log.txt file even when there was an error

Peter-St, 2020-05-16:
Please run this app with your endoscop camera:
https://drive.google.com/open?id=149-Bbiyj1WmHIJvzbd-H2nKTCfLHs2hr
increase the packetsPerRequest to 2 and the activeUrbs also to 2
Run the "Test The Stream" Button after Manual setup

Attach the log.txt file

ErnstTom, 2020-05-16:
previously as if I had read that the pen is not android capable.
although it recognizes it as usb capable. in time for the second

at the endoscope, I do the recommended data manually.

[appendosc2log.txt](https://github.com/Peter-St/Android-UVC-Camera/files/4638686/appendosc2log.txt)
[appendosclog.txt](https://github.com/Peter-St/Android-UVC-Camera/files/4638688/appendosclog.txt)
[penapp6log.txt](https://github.com/Peter-St/Android-UVC-Camera/files/4638689/penapp6log.txt)
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

Peter-St, 2020-05-16:
On the Pen Usb device only click the Manual Set Up button. After the error close the app and attach the log.txt file.

Peter-St, 2020-05-16:
https://drive.google.com/open?id=1mLBo1yZPJozDNxXTOMQ-mIjyFGLuSnNx
you can use this apk for your pen usb camera

ErnstTom, 2020-05-16:
does not do logtxt
with app6 (none). the error text is already generated in the pinat when i tap the uvc key. does not enter for manual or auto selection.
otherwise I tried to press the auto button earlier but got stuck at 0%

Peter-St, 2020-05-16:
Try to run all methods with the last app.

Peter-St, 2020-05-16:
Use the latest App and try it out again.

ErnstTom, 2020-05-16:
I change any data - which one is important, I don't know - but the test the stream always resets it.
and probe-commit always produces a null result.

Peter-St, 2020-05-16:
You do not need to change the entries. First we have to overcome this exception you get. Next you can set up the pen cam too... 

I don't know if we get this cameras with your phone to work.
Both cameras has a large Maxpacketsize with 3072 bytes, but I think you need a smaller one for your phone.
The endoscope cam could have one, but I don't know how to access it (1x768 bytes)

Peter-St, 2020-05-16:
When you get an output from the latest app with your pen camera, plz attach it.

Peter-St, 2020-05-16:
https://drive.google.com/open?id=1qrNB0_0ovxLG_Jt1Zev-COI_hNm_Cu1U
This app should write some outputs to the log.txt file when you run it with your pen camera.
Please try out all the buttons and then attach the log.txt file.
Also attach the log.txt file, when you get the Array Index Out of Bounds Exception.

[33 more replies]

closed 2021-10-01 by Peter-St