Peter-St/Android-UVC-Camera #12: do you know if it works for Device 004: ID 0bda:5806 Realtek Semiconductor Corp.

blassmegod, 2021-02-11:
Hi,

Do you know if it will work for the following device? I tried multiple configurations, but in the best case I'm getting only a black screen. On a Linux system it works great (guvcview: 30/1 fps, 640x480, YUYV 4:2:2) 

[lsusb_uvc.txt](https://github.com/Peter-St/Android-UVC-Camera/files/5965038/lsusb_uvc.txt)

Thanks,

Peter-St, 2021-02-12:
Hi,

plz attach the output of the following command from a linux terminal with you camera connected:

lsusb -v -d 0bda:5806

blassmegod, 2021-02-12:
Hi Peter,

I attached the output of the command. It's interesting that I get inside it also this message "Couldn't open device, some information will be missing"
[lsusb_outout.txt](https://github.com/Peter-St/Android-UVC-Camera/files/5970530/lsusb_outout.txt)

Peter-St, 2021-02-12:
Ok, start the app, connect the camera to your smartphone and eventually post a screenshot of the outputs you get from the buttons in the "Set Up The Camera Device" screen:

1st Button: "Find The Camera"
2nd Button: "List up The Camera"
3rd Button: "Set Up With UVC Settings"
`--> choose the Manual Method !`
     
     If you run into any errors in the  "Set Up With UVC Settings" button then post them.
1) MaxPacketSize: You select the highest Values in the List --> should be 3072
2) PacketsPerRequest: You select 2 for the first time.
3) ActiveUrbs: You select 2 for the first fime.
4) FormatIndex: You should only see one entry: YUY2 --> select it!
5) FrameIndex: Select the lowest Resolution: 320 x 240 ---> select it!
6) FrameInterval: Select 30 fps (333333)`

Save the values and post the output.
         
If everything works Click the Button "ControlTransfer - Testrun" and 1st click on "Video Probe-Commit Controls"
--> Post the output

Next Click on "ControlTransfer - Testrun" and select "Testrun" and the click on "One Frame"
--> Post the output

Next Click on "ControlTransfer - Testrun" and select "Testrun" and the click on "Frames for 5 secounds"
--> Post the output

If you get an error, then post in which step you get the error.

So far,

Peter

blassmegod, 2021-02-14:
It almost works, there are a few rows from the image output. I attached what you requested. Thank you
[image][image][image][image][image][image]

Peter-St, 2021-02-14:
ok, so let me explain you something:

[image]

Here you see, that the VideoFrameSize should be: 153600 bytes
This means your frames should all be 153600 bytes long!
The frames you receive are only 600 to 700 bytes long!

You first have to run the testruns so long, until you receive long enough frames.
So you can try to increase the avtiveUrbs for example.
Or you can increase the packetsPerRequest.
Try out differnt values for this two variables:
For example: packtesPerRequest: 4, activeUrbs: 32
You can edit this two Values by clicking on "Edit / Save The Camera Values" --> edit only the PacketsPerRequest and ActiveUrbs entries!
--> Save the Values
-- > reRun the Testrun and post the output.

blassmegod, 2021-02-15:
I haven't yet found the right values for ActiveUrbs and PacketsPerRequest, but I have two other questions:
- why do you reset activeUrbs to value 16 in five places from libusb_support.c: `if (activeUrbs > 16) activeUrbs = 16;`
- if `activeUrbs = maxPacketSize x packetsPerRequest`, then it's value should be quite big even for a packet per request as in my case maxPacketSize=3072 bytes, or should the maxPacketSize be converted first to KB or MB.....?

Peter-St, 2021-02-15:
The active Urbs normally does not need to be reset to 16. I think that was originally because of an automatic setup, which should not count over 16.

The value is allways in bytes. You can also try to lower the MaxPacketSize to 1028  (depends on the Altsetting) or event. to 512. --> Then the Packets arrive faster, because their size is smaller. All in all this values were Device Specific. They depends on the usb camera on the one side and on the smartphone on the other. You can try to set them up by yourself with this app, or use a 3rd party app with fix values (can lead to malfunctions on some devices).

So far,

Peter

closed 2021-10-01 by Peter-St