Peter-St/Android-UVC-Camera #13: Can't compile the project

ToysoftInc, 2021-02-13:
Execution failed for task ':app:externalNativeBuildCleanCustomDebugType'.
> A problem occurred starting process 'command '/home/peter/Android/Sdk/ndk/android-ndk-r15c/ndk-build''

Peter-St, 2021-02-14:
Hello ToysoftInc,

you have to set up this Android Studio Project on your own PC. I'm using Linux to compile this project.

> > A problem occurred starting process 'command '/home/peter/Android/Sdk/ndk/android-ndk-r15c/ndk-build''

--> It points to a path, which can only work on my PC, because this is the ndk path for only my PC.

- Edit this file: local.properties    --> reset the sdk and nkd paths, so they fit to your location.

Note: To compile the project, I'm using NDK verison: 21.1.6352462   . If i use an older or a newer one, the build could also fail.

Good Luck,

Peter

ToysoftInc, 2021-02-14:
Do you think you can release a lib version?
Also would be help to have some description about this project. :)

Thanks
Danny

Peter-St, 2021-02-14:
Ok, I reupdated the readme file.

If you face problems in compiling the native part, then you also can skip compiling the native libs by excluding the following in the build.gradle file:

externalNativeBuild {
        ndkBuild {
            path 'src/main/jni/Android.mk'
        }
    }

habi-kvana, 2021-02-17:
iam also facing compilation issue because of ndk path

This is my ndk path —>ndk.dir=/Users/dolly/Library/Android/sdk/ndk/21.1.6352462

Execution failed for task ':app:ndkBuild'.
> Process 'command '/Users/dolly/Library/Android/sdk/ndk/21.1.6352462/ndk-build'' finished with non-zero exit value 2

Peter-St, 2021-02-17:
> iam also facing compilation issue because of ndk path
> 
> This is my ndk path —>ndk.dir=/Users/dolly/Library/Android/sdk/ndk/21.1.6352462
> 
> Execution failed for task ':app:ndkBuild'.
> 
> > Process 'command '/Users/dolly/Library/Android/sdk/ndk/21.1.6352462/ndk-build'' finished with non-zero exit value 2

I had the same issue, it worked with an older ndk tool:
Android NDK, Revision 20b
https://developer.android.com/ndk/downloads/older_releases

Try it to compile it with this version: 20.1.5948944

Otherwise look at the ndk error message you get in Android-Studio and try to find a solution.

Also deleting the build folder and running Invalidate Caches / Restart sometimes helps.

If you find the Error passage in the build log, then you can also post it here, because your error message is the overview error message.

ToysoftInc, 2021-02-18:
Peter, do you think you can create a lib for us?

Peter-St, 2021-02-18:
I would suggest, if you don't want to edit the native code, to skip the ndk part, because the libs already are precompiled in the libs folder.

For an android AAR Library you would also have to run the ndk part, which I think could also fail then.

So far,

Peter

Peter-St, 2023-03-02:
Build System changed from Autotools to Cmake.

Perhaps you can give em another try.

closed 2023-03-02 by Peter-St