Peter-St/Android-UVC-Camera #17: Unable to see connected external usb cameras in android mobile browser, Will mobile browser detects an external usb camera like native cameras?

SreeniMannem, 2021-02-24:
We are unable to see the connected external Usb camera in android mobile browser. The same piece of code will detect and show the connected external Usb cameras in desktop browsers.

Is there anything wrong with the detecting external USB cameras in mobile browsers? will mobile browser supports it or not?

**Note:** We are using "`navigator.mediaDevices.enumerateDevices()`" to list the available and connected devices in the browser.

Peter-St, 2021-02-24:
Hello,

read this:

https://stackoverflow.com/questions/60297972/navigator-mediadevices-enumeratedevices-returns-empty-labels

SreeniMannem, 2021-03-03:
Thanks for your help Peter, this was working with desktop browser. But i need help on how to get connected external devices on Mobile web browsers. Kindly do needful.

Peter-St, 2021-03-06:
Sorry, but I think I can't help you, because I haven't any experience with mobile browsers up to now.

Peter

closed 2021-10-01 by Peter-St