Peter-St/Android-UVC-Camera #19: Preview not displaying for external camera

habi-kvana, 2021-03-23:
Preview is not displaying for external camera product-id: 102, vendor-id" 4057
[image]

Peter-St, 2021-03-28:
Hello,

you have to find out, if the camera is an uvc device:
On a Linux System run this command:

lsusb -v -d xxxx:xxxx
The first 4 x are the product and the second 4 x are the vendor id.
If you can run this command successfully, you can attach the output.

Thanks, Peter

habi-kvana, 2021-03-30:
thanks for responding @Peter-St 
I resolved this issue after opening the camera we should get SupportedSize and then pass one of those resolutions then it will display preview.

closed 2021-03-30 by Peter-St