Peter-St/Android-UVC-Camera #20: Pure Termal Issue

Peter-St, 2021-03-30:
> @ 38293312可以将纯热固件升级到版本1.3.0，然后重试吗？
> https://github.com/groupgets/purethermal1-firmware/releases
Yes, I upgraded the firmware of PT2, but the phone still can’t use it normally
Please watch the test video： https://www.youtube.com/watch?v=3SFKlTRe-G0
[image]

_Originally posted by @38293312 in https://github.com/Peter-St/Android-UVC-Camera/issues/18#issuecomment-809992625_

Peter-St, 2021-03-30:
> > @ 38293312嗯，这似乎不是应用程序问题。它甚至没有检测到纯热板。我建议尝试使用其他电缆以确保其USB OTG。一旦您连接pt2板，Android甚至应该在启动该应用之前对其进行检测。
> 
> It can be checked that PT2 is UVC. Just click the default option to open it with the APP, just like disconnecting the UVC from the mobile phone. For the same mobile phone, I can use another OTG thermal imaging camera normally with OTG

Peter-St, 2021-03-30:
> This is the parameter of "PT2 + LEPTON3.5" viewed from the PC software "Lepton User App"
> 
> [image]
> [image]
> 
> [image]
> [image]
> 
> [image]
> [image]

Peter-St, 2021-03-30:
> I use PT2+LEPTON3.5, APP "UYVY_support.apk" test. Failed!
> 
> [image]
> [image]
> [image]
> [image]
> [image]
> 
> After the above problem occurs, unplug the OTG and reconnect it, the following content is displayed
> [image]

Peter-St, 2021-03-30:
> @38293312 May you upgrade the purethermal firmware to version 1.3.0 and try again?
> https://github.com/groupgets/purethermal1-firmware/releases

38293312, 2021-03-31:
I upgraded the PT2 to the 1.3 firmware, and it failed. Here is the test video: https://www.youtube.com/watch?v=3SFKlTRe-G0&t=15s

closed 2021-10-01 by Peter-St