Peter-St/Android-UVC-Camera #23: can give more priority to usb camer on android device using this library

EworkitSrl, 2021-10-04:
i want to develop an app that when i connect a third party camera to device  then the device should take that camera as the main camera.
ex-: if  get a video call in any application, the camera view should take from that third party camera. Is this possible?

Peter-St, 2021-10-04:
Hello EworkitSrl,

the Google Developer Team dropped with Android Pie (SDK28) the Usb camera support ([https://github.com/saki4510t/UVCPermissionTest](url)).
Please read through this link and also read the open issiue in this link.

Theoretically it could only work on older Android devices.
Another thing is, that with the upcomming USB3 speed, UVC is no more a common driver way for usb cameras. The google team started to include an usb camera support in Android Pie , but never finished it in all ongoing releases.

So far,
Peter

heaveninfosoft, 2021-12-23:
https://play.google.com/store/apps/details?id=com.shenyaocn.android.usbcamera
how this app work in all device??
@Peter-St @EworkitSrl

closed 2023-03-23 by Peter-St