Peter-St/Android-UVC-Camera #25: Preview will freeze

manxialiunian, 2021-11-15:
Hello, we are pulling the code to compile and install it on the mobile phone, and the external USB camera, the preview is all OK; but when the USB camera is facing the dense square cloth, the phenomenon of freezing will occur, please see what the problem is

https://user-images.githubusercontent.com/15249057/141757308-3f423c35-d98f-448f-b562-204372466221.mp4

Peter-St, 2021-11-16:
Hello manxialiunian,

1) you can try to increase or decrease the ActiveUrbs or PacketsPerRequest Value to influenze the output.

2) In my app the image processing program code needs to be improved, because sometimes there were also broken frames in the stream, which happens due to a weak program code. I'm workin on it.

manxialiunian, 2021-11-17:
> Hello manxialiunian,
>
> 1. you can try to increase or decrease the ActiveUrbs or PacketsPerRequest Value to influenze the output.
> 2. In my app the image processing program code needs to be improved, [...]

closed 2021-11-17 by manxialiunian

manxialiunian, 2021-11-17:
thank you for your reply. By adjusting ActiveUrbs or PacketsPerRequest, the problem still occurs at 1080p resolution.