Peter-St/Android-UVC-Camera #26: WebRTC crashing

lchandp, 2021-11-24:
Compiled the source and uvc camera working on Android 9. 

But testing the webrtc, the device is crashing and throwing the following message

11-24 07:11:37.689  6366  6618 D From Native: UVC_STREAM_ERR --> Package 12
11-24 07:11:37.689  6366  6618 D From Native: UVC_STREAM_ERR --> Package 13
11-24 07:11:37.689  6366  6618 F libc    : Fatal signal 11 (SIGSEGV), code 1 (SEGV_MAPERR), fault addr 0x3 in tid 6618 (IntentService[W), pid 6366 (umer.uvc_camera)
11-24 07:11:37.811  6629  6629 I crash_dump32: obtaining output fd from tombstoned, type: kDebuggerdTombstone
11-24 07:11:37.812  3272  3272 I /system/bin/tombstoned: received crash request for pid 6618
11-24 07:11:37.812  6629  6629 I crash_dump32: performing dump of process 6366 (target tid = 6618)
11-24 07:11:37.849  6629  6629 F DEBUG   : *** *** *** *** *** *** *** *** *** *** *** *** *** *** *** ***
11-24 07:11:37.849  6629  6629 F DEBUG   : Build fingerprint: 'Amlogic/franklin/franklin:9/PPR1.180610.011/20210913:userdebug/test-keys'
11-24 07:11:37.849  6629  6629 F DEBUG   : Revision: '0'
11-24 07:11:37.849  6629  6629 F DEBUG   : ABI: 'arm'
11-24 07:11:37.849  6629  6629 F DEBUG   : pid: 6366, tid: 6618, name: IntentService[W  >>> humer.uvc_camera <<<
11-24 07:11:37.849  6629  6629 F DEBUG   : signal 11 (SIGSEGV), code 1 (SEGV_MAPERR), fault addr 0x3
11-24 07:11:37.849  6629  6629 F DEBUG   : Cause: null pointer dereference
11-24 07:11:37.849  6629  6629 F DEBUG   :     r0  0000008d  r1  00000020  r2  9a9fa5f9  r3  00000003
11-24 07:11:37.849  6629  6629 F DEBUG   :     r4  93ab624c  r5  0000000e  r6  93ab6324  r7  0000000c
11-24 07:11:37.849  6629  6629 F DEBUG   :     r8  00000000  r9  926b1040  r10 8eb27400  r11 8ee8ade8
11-24 07:11:37.849  6629  6629 F DEBUG   :     ip  926b1014  sp  8ee8adc0  lr  926aabe0  pc  926aac30
11-24 07:11:38.075  6629  6629 F DEBUG   : 
11-24 07:11:38.075  6629  6629 F DEBUG   : backtrace:
11-24 07:11:38.075  6629  6629 F DEBUG   :     #00 pc 0000ec30  /data/app/humer.uvc_camera-LwypGrpriOEug48kSJD0PA==/lib/arm/libUvc_Support.so (cb_jni_WebRtc_Service+204)
11-24 07:11:38.075  6629  6629 F DEBUG   :     #01 pc 00006f11  /data/app/humer.uvc_camera-LwypGrpriOEug48kSJD0PA==/lib/arm/libusb1.0.so
11-24 07:11:38.075  6629  6629 F DEBUG   :     #02 pc 0000ad97  /data/app/humer.uvc_camera-LwypGrpriOEug48kSJD0PA==/lib/arm/libusb1.0.so
11-24 07:11:38.075  6629  6629 F DEBUG   :     #03 pc 00009b4f  /data/app/humer.uvc_camera-LwypGrpriOEug48kSJD0PA==/lib/arm/libusb1.0.so
11-24 07:11:38.075  6629  6629 F DEBUG   :     #04 pc 0000773b  /data/app/humer.uvc_camera-LwypGrpriOEug48kSJD0PA==/lib/arm/libusb1.0.so
11-24 07:11:38.075  6629  6629 F DEBUG   :     #05 pc 00007333  /data/app/humer.uvc_camera-LwypGrpriOEug48kSJD0PA==/lib/arm/libusb1.0.so (libusb_handle_events_timeout_completed+238)
11-24 07:11:38.075  6629  6629 F DEBUG   :     #06 pc 000077bf  /data/app/humer.uvc_camera-LwypGrpriOEug48kSJD0PA==/lib/arm/libusb1.0.so (libusb_handle_events+16)
11-24 07:11:38.075  6629  6629 F DEBUG   :     #07 pc 0000f264  /data/app/humer.uvc_camera-LwypGrpriOEug48kSJD0PA==/lib/arm/libUvc_Support.so (lunchTheStream_WebRtc_Service+524)
11-24 07:11:38.075  6629  6629 F DEBUG   :     #08 pc 00010ce0  /data/app/humer.uvc_camera-LwypGrpriOEug48kSJD0PA==/lib/arm/libjnidispatch.so (ffi_call_SYSV+44)
11-24 07:11:38.075  6629  6629 F DEBUG   :     #09 pc 00010840  /data/app/humer.uvc_camera-LwypGrpriOEug48kSJD0PA==/lib/arm/libjnidispatch.so (ffi_call+304)
11-24 07:11:38.075  6629  6629 F DEBUG   :     #10 pc 000057b0  /data/app/humer.uvc_camera-LwypGrpriOEug48kSJD0PA==/lib/arm/libjnidispatch.so
11-24 07:11:38.075  6629  6629 F DEBUG   :     #11 pc 000079b4  /data/app/humer.uvc_camera-LwypGrpriOEug48kSJD0PA==/lib/arm/libjnidispatch.so (Java_com_sun_jna_Native_invokeVoid+44)
11-24 07:11:38.075  6629  6629 F DEBUG   :     #12 pc 0041c279  /system/lib/libart.so (art_quick_generic_jni_trampoline+40)
11-24 07:11:38.075  6629  6629 F DEBUG   :     #13 pc 00417d75  /system/lib/libart.so (art_quick_invoke_stub_internal+68)
11-24 07:11:38.075  6629  6629 F DEBUG   :     #14 pc 003f13eb  /system/lib/libart.so (art_quick_invoke_static_stub+222)
11-24 07:11:38.075  6629  6629 F DEBUG   :     #15 pc 000a1043  /system/lib/libart.so (art::ArtMethod::Invoke(art::Thread*, unsigned int*, unsigned int, art::JValue*, char const*)+154)
11-24 07:11:38.075  6629  6629 F DEBUG   :     #16 pc 001e8835  /system/lib/libart.so (art::interpreter::ArtInterpreterToCompiledCodeBridge(art::Thread*, art::ArtMethod*, art::ShadowFrame*, unsigned short, art::JValue*)+232)
11-24 07:11:38.075  6629  6629 F DEBUG   :     #17 pc 001e3511  /system/lib/libart.so (bool art::interpreter::DoCall<false, false>(art::ArtMethod*, art::Thread*, art::ShadowFrame&, art::Instruction const*, unsigned short, art::JValue*)+776)
11-24 07:11:38.075  6629  6629 F DEBUG   :     #18 pc 003eceeb  /system/lib/libart.so (MterpInvokeStatic+130)
11-24 07:11:38.076  6629  6629 F DEBUG   :     #19 pc 0040ab94  /system/lib/libart.so (ExecuteMterpImpl+14612)
11-24 07:11:38.076  6629  6629 F DEBUG   :     #20 pc 00267284  /dev/ashmem/dalvik-classes.dex extracted in memory from /data/app/humer.uvc_camera-LwypGrpriOEug48kSJD0PA==/base.apk (deleted) (com.sun.jna.Function.invoke+960)
11-24 07:11:38.076  6629  6629 F DEBUG   :     #21 pc 001c7e89  /system/lib/libart.so (_ZN3art11interpreterL7ExecuteEPNS_6ThreadERKNS_20CodeItemDataAccessorERNS_11ShadowFrameENS_6JValueEb.llvm.2193211614+352)
11-24 07:11:38.076  6629  6629 F DEBUG   :     #22 pc 001cc757  /system/lib/libart.so (art::interpreter::ArtInterpreterToInterpreterBridge(art::Thread*, art::CodeItemDataAccessor const&, art::ShadowFrame*, art::JValue*)+146)
11-24 07:11:38.076  6629  6629 F DEBUG   :     #23 pc 001e34fb  /system/lib/libart.so (bool art::interpreter::DoCall<false, false>(art::ArtMethod*, art::Thread*, art::ShadowFrame&, art::Instruction const*, unsigned short, art::JValue*)+754)
11-24 07:11:38.076  6629  6629 F DEBUG   :     #24 pc 003ebf0f  /system/lib/libart.so (MterpInvokeVirtual+442)
11-24 07:11:38.076  6629  6629 F DEBUG   :     #25 pc 0040aa14  /system/lib/libart.so (ExecuteMterpImpl+14228)
11-24 07:11:38.076  6629  6629 F DEBUG   :     #26 pc 002673fc  /dev/ashmem/dalvik-classes.dex extracted in memory from /data/app/humer.uvc_camera-LwypGrpriOEug48kSJD0PA==/base.apk (deleted) (com.sun.jna.Function.invoke+348)
11-24 07:11:38.076  6629  6629 F DEBUG   :     #27 pc 001c7e89  /system/lib/libart.so (_ZN3art11interpreterL7ExecuteEPNS_6ThreadERKNS_20CodeItemDataAccessorERNS_11ShadowFrameENS_6JValueEb.llvm.2193211614+352)
11-24 07:11:38.076  6629  6629 F DEBUG   :     #28 pc 001cc757  /system/lib/libart.so (art::interpreter::ArtInterpreterToInterpreterBridge(art::Thread*, art::CodeItemDataAccessor const&, art::ShadowFrame*, art::JValue*)+146)
11-24 07:11:38.076  6629  6629 F DEBUG   :     #29 pc 001e4377  /system/lib/libart.so (bool art::interpreter::DoCall<true, false>(art::ArtMethod*, art::Thread*, art::ShadowFrame&, art::Instruction const*, unsigned short, art::JValue*)+674)
11-24 07:11:38.076  6629  6629 F DEBUG   :     #30 pc 003ed149  /system/lib/libart.so (MterpInvokeVirtualRange+424)
11-24 07:11:38.076  6629  6629 F DEBUG   :     #31 pc 0040ad14  /system/lib/libart.so (ExecuteMterpImpl+14996)
11-24 07:11:38.076  6629  6629 F DEBUG   :     #32 pc 00268420  /dev/ashmem/dalvik-classes.dex extracted in memory from /data/app/humer.uvc_camera-LwypGrpriOEug48kSJD0PA==/base.apk (deleted) (com.sun.jna.Library$Handler.invoke+520)
11-24 07:11:38.076  6629  6629 F DEBUG   :     #33 pc 001c7e89  /system/lib/libart.so (_ZN3art11interpreterL7ExecuteEPNS_6ThreadERKNS_20CodeItemDataAccessorERNS_11ShadowFrameENS_6JValueEb.llvm.2193211614+352)
11-24 07:11:38.076  6629  6629 F DEBUG   :     #34 pc 001cc757  /system/lib/libart.so (art::interpreter::ArtInterpreterToInterpreterBridge(art::Thread*, art::CodeItemDataAccessor const&, art::ShadowFrame*, art::JValue*)+146)
11-24 07:11:38.076  6629  6629 F DEBUG   :     #35 pc 001e34fb  /system/lib/libart.so (bool art::interpreter::DoCall<false, false>(art::ArtMethod*, art::Thread*, art::ShadowFrame&, art::Instruction const*, unsigned short, art::JValue*)+754)
11-24 07:11:38.077  6629  6629 F DEBUG   :     #36 pc 003ecadd  /system/lib/libart.so (MterpInvokeInterface+1020)
11-24 07:11:38.077  6629  6629 F DEBUG   :     #37 pc 0040ac14  /system/lib/libart.so (ExecuteMterpImpl+14740)
[... cut ...]
11-24 07:11:39.217  3463  6644 W ActivityManager:   Force finishing activity humer.uvc_camera/com.example.androidthings.videortc.WebRtc_MainActivity
11-24 07:11:39.239  3463  6644 W ActivityManager:   Force finishing activity humer.uvc_camera/com.example.androidthings.videortc.CallActivity
11-24 07:11:39.251  3463  3478 W BroadcastQueue: Background execution not allowed: receiving Intent { act=android.intent.action.DROPBOX_ENTRY_ADDED flg=0x10 (has extras) } to com.google.android.gms/.stats.service.DropBoxEntryAddedReceiver

Peter-St, 2021-11-24:
Hi,

I noticed the crash on WeRtc. There wil be major changes when using libusb with the camera and after this, I will rewrite the webrtc progress.

So far,

Peter

Peter-St, 2023-03-02:
Hi Again,

I solved everything, but by the while AppRTC isn't running any more.

I decided to dismiss the WebRTC support

closed 2023-03-02 by Peter-St