Peter-St/Android-UVC-Camera #27: Unable To See Preview

mehmetakify, 2022-01-20:
After set targetSdkVersion to 27, in Samsung Tab A7 (Android 11) and Vestel Venus V7 (Android 9) I can not see preview. There is only a grey screen.

[image]

[image]
[image]
[image]

Peter-St, 2022-01-20:
Hello  mehmetakify and thx for commenting.

I'm doing a major change in processing the stream in my app. The changes are not complete, so actually such things could happen. I will continue at the weekend and if everything works, I will post it to you.

So far,

Peter

mehmetakify, 2022-01-20:
Hello Peter,

If you will achieve to solve the problem such a short time you will be my hero.

Thanks for your kind response.

Sumeyya-Eser, 2022-08-09:
> Hello Peter,
> 
> If you will achieve to solve the problem such a short time you will be my hero.
> 
> Thanks for your kind response.

Hello Mehmet Akif, do you have any update ?

mehmetakify, 2022-08-09:
Hello Sumeyye unfortunately not. I forced my customer to use one of the natively supported device. There are several types.

Peter-St, 2022-08-10:
Hi,

can you plz try my app again. Compile it with the latest commit and try if you can catch a picture.