Peter-St/Android-UVC-Camera #28: Video Probe-Commit Control failed!

ZhiminWang96, 2022-03-03:
Hi !
I downloaded the app from the google play.
I tried to set different parameters. One of these is shown as follows.
[image]
However, when I click the "Video Probe-Commit Control" button, the message always shows below.
[image]
I don't know why the 'Camera Frame Index' from the Camera Controltransfer is always equal to 0.
I will be appreciate it if you can help me.
Thank you very much for your contribution.
Some information is as follows.
[image]

[image]

[image]

Peter-St, 2022-03-03:
Hello ZhiminWang96,

it seems to be, that there is a failure in my app.

I will update my app, than you can try it again, but it will take some time.

So far, Peter

Peter-St, 2023-03-23:
Hi,

WebRTC is sadly discontinued, so I dropped the WebRTC support.

So far,

Peter

closed 2023-03-23 by Peter-St