Peter-St/Android-UVC-Camera #30: infrared UVC not showing

heaveninfosoft, 2022-05-24:
infrared UVC not showing

Peter-St, 2022-05-29:
Hi,

I have updated the repo with UYVY support. Pls try out the repo again and report your results.

Thx, Peter

heaveninfosoft, 2022-05-31:
why ndk error show? it not occur in earlier version

Peter-St, 2022-05-31:
I'm going to solve this issue.

It's because I included the libuvc library, but something isn't working as it should.

Peter-St, 2022-05-31:
Hi,

can you plz attach the log from the ndk build failure.
Plz tell, which version of the ndk build tools you use.

heaveninfosoft, 2022-05-31:
Caused by: com.android.builder.errors.EvalIssueException: [CXX1405] error when building with ndkBuild using B:\XXX\Android-UVC-Camera-master\Android-UVC-Camera-master\app\src\main\jni\Android.mk: Build command failed.

Caused by: org.gradle.process.internal.ExecException: Process 'command 'E:\XXXX\24.0.8215888/ndk-build.cmd'' finished with non-zero exit value 2

heaveninfosoft, 2022-05-31:
i think  same happen because i tried too on that version earlier

Peter-St, 2022-05-31:
Hm,
change your apps build gradle file to your ndk build version.
Try it out again with the version "21.1.6352462" and attach the log.
Search through the log in Android Studio and post all errors.

Thx, Peter

Peter-St, 2022-06-05:
Hi,

I changed the native Build System from NDK to CMake.

You can try it again now.