Peter-St/Android-UVC-Camera #31: can i use this app for my camera PLUG417

mumairali, 2022-07-04:
i install it and it shows very bad feed on my app help me

Peter-St, 2022-07-05:
Hello,

it can be, that this camera works with my app.

Try to find out, if the camera is an UVC device:

> On a Linux System run this command:

> lsusb` -v -d xxxx:xxxx
> The first 4 x are the product and the second 4 x are the vendor id.
> If you can run this command successfully, you can attach the output.

If the camera supports uvc, you can use it with my app.

mumairali, 2022-07-05:
bro check this please 
[USB board user manual-V1.1-20200316.pdf](https://github.com/Peter-St/Android-UVC-Camera/files/9048923/USB.board.user.manual-V1.1-20200316.pdf)

mumairali, 2022-07-05:
this file has all the information

mumairali, 2022-07-05:
also can i get the y16 data in your app?

mumairali, 2022-07-05:
also I try with your app but its results not good see 
[camera output.txt](https://github.com/Peter-St/Android-UVC-Camera/files/9048963/camera.output.txt)

mumairali, 2022-07-05:
also i have test video 

https://user-images.githubusercontent.com/74451202/177401391-05cf6e05-bbe1-4677-8b2f-3d719d7e17ea.mp4

mumairali, 2022-07-05:
also i think it needs to add bulk end points 
https://linux-usb.vger.kernel.narkive.com/Z1eSNhTV/patch-1-1-usb-gadget-uvc-add-support-for-bulk-endpoint-to-be-used-as-video-streaming-ep

Peter-St, 2022-07-05:
ok,

i noticed you used uyvy as video format, but the usb adapter supports yuy2 and y16 format.

Try to change to yuy2 format and see if it works.

To know if it supports bulk endpoints, you have to read out the uvc descriptor with the command from my first post and attach the output.

Generaly you are able to receive the frames, but the video conversation from my app fails. 
I have recently made some updates to my app, but had not enough time, to test them, so some bugs could happen too.

Thx, Peter

mumairali, 2022-07-06:
idVendor           0x04b4 Cypress Semiconductor Corp.
idProduct          0xf8f8

mumairali, 2022-07-07:
here i try using your settings https://user-images.githubusercontent.com/74451202/177695994-249330f6-733e-4770-b90e-985b2134fe4e.MOV

mumairali, 2022-07-07:
here it work with these settings https://user-images.githubusercontent.com/74451202/177696036-21ba9211-40ca-473e-9a2c-8a94aec0cb9b.MOV

Peter-St, 2022-07-08:
Hi

- Can you plz make a video from the button "Set Up With Uvc Setting"     --> choose the manual method and select all inputs.
-  It looks like your camera supports bulk transfer.
- It would also help, if you can read out the UVC informations on a linux PC:

lsusb` -v -d xxxx:xxxx
The first 4 x are the product and the second 4 x are the vendor id.
If you can run this command successfully, you can attach the output.

Peter-St, 2023-03-23:
Hi,

Your camera should run with my app.
Bulk endpoint stream is enabled in the latest version. I have tested several Usb 3.0 UVC cams which works fine.

On the video you posted, the picture you got looks like it was not correctly converted .
Perhaps you can try out the app with the latest changes, so if this frame conversation could be fixed.