Peter-St/Android-UVC-Camera #32: NV12 Video Format not Supported

OmaGamer, 2022-11-16:
Hey, I tried to run the PSVita on this program. On the other programs, it works just fine.
[image]
[image]

Peter-St, 2022-11-18:
Hi and thx for trying out my app.

First another thing --> don't run the program with 0 packetsPerRequest --> change this value at least to 1 or more.

Next for the camera format. Sadly NV12 is not supported by now. If I get some time in the near future, I will try to include the NV12 Format and will notice it to you in this forum.

Thx, Peter

OmaGamer, 2022-11-18:
Thanks for the response!
I tried some packetsPerRequest values and my UCV device did recognize the phone this time. Sadly I just got a black screen. I guess this is because my device only supports N12.

Thanks for the great app, it is the only free option.

Peter-St, 2022-11-19:
Perhaps try out the other video formats. Perhaps you can get a result there.