Peter-St/Android-UVC-Camera #34: Support to stream onboard camera data to PC

sooorajjj, 2023-05-21:
Hi @Peter-St, is it possible to stream android devices onboard camera data to PC using USB UVC gadget function using your libuvc?
I have custom android firmware for Pixel 4a where I have enabled UVC gadget function on kernel side , which creates a video device node on device side /dev/video3 three , was looking for a JNI implementation which allows me to stream camera data to PC via USB.

This means I can use my device as USB webcam on PC. 

Looking forwarder to your response 
Thanks

Peter-St, 2023-05-22:
> Hi @Peter-St, is it possible to stream android devices onboard camera data to PC using USB UVC gadget function using your libuvc? I have custom android firmware for Pixel 4a where I have enabled UVC [...]

Hi,

I read the documentation and think that it should be possible.

One thing bothering me:
Your Phone only has one usb otg port, which needs to be connected to your pc, to get the USB UVC gadget  working. You will need at least one more usb port, to connect the usb camera. The questions is, if the USB UVC gadget also works from an usb hub. (/dev/video3 tree get created on you pc --> with pc and camera connected to the hub.)
Can you plz check it out.

sooorajjj, 2023-05-22:
> Hi,
>
> I read the documentation and think that it should be possible.
>
> One thing bothering me: Your Phone only has one usb otg port, which needs to be connected to your pc, to get the USB UVC gadget [...]

What I mean is use camera on android device to stream camera frames to PC , since we are using the built in camera on the device we only need one USB port to connect to PC ...
And the debugging can be using ADB over wifi .

Peter-St, 2023-05-22:
Oh, ok. 
I think it should work. 
You will need the souce code of an app, which catches the frames from the internal camera and next you will have to create a userspace program to interact with the UVC Gadget driver.

`The UVC Gadget driver must be paired with a userspace program that responds to UVC control requests and fills buffers to be queued to the V4L2 device that the driver creates. How those things are achieved is implementation dependent and beyond the scope of this document, but a reference application can be found at https://gitlab.freedesktop.org/camera/uvc-gadget`
--> https://lwn.net/Articles/925750/

I have an idea how this app should look like, but I don't own an Android device with custom kernel source, so I think, I can't help you out in this issue.
My app in general is only for usb cameras. Your userspace program will have to catch the camera frames in java.  Next you will have to pass them to C / C++ code and then I think you will have to interact with your pc using the IOCTL's from the V4L2 library.       (The v4l2.c file from the sample link above shows this approach). 
This are only thoughts. To be more specific, I would need an android dev with custom kernel.

So far,

Peter

sooorajjj, 2023-05-22:
> Oh, ok. I think it should work. You will need the souce code of an app, which catches the frames from the internal camera and next you will have to create a userspace program to interact with the [...]

Just sent you a mail , please let me know if you have not received it.