Peter-St/Android-UVC-Camera #38: Request: Create libuvc separate fork

nsknyc, 2023-12-21:
Hello Peter,
I noticed that the libuvc library you have included in the repo has a readme to ktossell github. I checked and that seems to be the old name. Anyways I was wondering if it's possible for you to fork libuvc to keep track of changes. The reason I ask is that groupgets (purethermal) also has some modifications to libuvc, and I'd hate to cause some conflicts. I also don't want to replace your libuvc version with a potentially older version (when I do pull requests). With a fork I can compare the official repo, with your modifications and merge groupgets' as well.

I hope I wasn't too confusing. I have radiometric data working on a test app with Y16 and would like to implement it here. I don't have much experience with other thermal cameras, but I can assume that they function the same way. Maybe this addition will help other.

Thanks!

Peter-St, 2023-12-24:
Hi  nsknyc,

The libuvc version in this Repo is modified 2 times. First it was modified to work with an older LibUsb versions and working with Android. After that I modified the version again to work with my custom driver input.
Replacing the libuvc version of this repo with an official fork of the libuvc version will follow in a lot of overlaps in the code. It`s even hard to solve all the overlaps "I" commited to this version. Then all the previous commits needs to be merged too, which I think I`m not aware of.
I think at this point it would be easier to create a new app based on the latest libuvc version to get it running, before getting this merged version sucessfully running with the latest libuvc version.
Another thing would be, that differnet upgrads in my app and in the official libuvc version will lead to a lot of conflicts during time. I think I will spare this work to me.

So far, upgrades in the official LibUvc version can also be commited to this libuvc version relativ easily, because the two libuvc versions are still quite related. If you have some suggestions, I`m quite interested in hearing them.

Thx, Peter

nsknyc, 2023-12-24:
Understood! I'll bring over the modifications (from another fork) over to this one. This way we can have radiometric data in this app. 
"If you have some suggestions, I`m quite interested in hearing them."
As far as the lib, that was pretty much it. Any modifications to the app itself I'll just test in another branch.

Thanks.

closed 2023-12-28 by Peter-St