Peter-St/Android-UVC-Camera #40: Crash on setting manual settings

ben-dempsey, 2023-12-29:
I am seeing a new crash when attempting to set the manual settings for my camera.

I am also unable to pull the latest commit because of an invalid path, it looks like there is a file ending with "." instead of ".jpeg"

I am doing the following:
- Plug in camera and open the app (Permissions already allowed)
- Click _SET UP THE CAMERA DEVICE_
- Click _FIND THE CAMERA_ (successful)
- Click _LIST UP THE CAMERA_ (successful)
- Click _SET UP WITH UVC SETTINGS_
- Click _MANUAL_ on dialog box
 **App Crash**
 
 Here is the logcat from where the crash occurs (removed timestamps for readability):

```
from_LibUVC             humer.uvc_camera                     D  _prepare_stream_ctrl sucessfully
from_LibUVC             humer.uvc_camera                     E  _uvc_get_stream_ctrl_format failed
from_LibUVC             humer.uvc_camera                     D  _uvc_get_stream_ctrl_format failed: -51
From libuvc_support     humer.uvc_camera                     E  could not negotiate with camera:err=-51
 From UVCPreviw          humer.uvc_camera                     D  end (-51)
libc                    humer.uvc_camera                     A  Fatal signal 11 (SIGSEGV), code 1 (SEGV_MAPERR), fault addr 0x42bbcf9842bbcf98 in tid 28752 (umer.uvc_camera), pid 28696 (umer.uvc_camera)
WindowManager           system_server                        E  win=Window{4a45d90 u0 humer.uvc_camera/humer.UvcCamera.SetUpTheUsbDeviceUvc EXITING} destroySurfaces: appStopped=false cleanupOnResume=false win.mWindowRemovalAllowed=true win.mRemoveOnExit=true win.mViewVisibility=0 caller=com.android.server.wm.WindowState.onExitAnimationDone:5997 com.android.server.wm.WindowStateAnimator.onAnimationFinished:225 com.android.server.wm.WindowState.onAnimationFinished:6222 com.android.server.wm.WindowContainer$$ExternalSyntheticLambda4.onAnimationFinished:2 com.android.server.wm.SurfaceAnimator.lambda$getFinishedCallback$0:140 com.android.server.wm.SurfaceAnimator.$r8$lambda$lRxTVOJy8fX752UbrFno9INW9hE:0 com.android.server.wm.SurfaceAnimator$$ExternalSyntheticLambda1.run:8 
DEBUG                   crash_dump64                         A  Cmdline: humer.uvc_camera
DEBUG                   crash_dump64                         A  pid: 28696, tid: 28752, name: umer.uvc_camera  >>> humer.uvc_camera <<<
DEBUG                   crash_dump64                         A        #05 pc 000000000001b48c  /data/app/~~<removed>==/humer.uvc_camera-px-<removed>==/lib/arm64/libuvc_preview.so (UVCPreview::preview_thread_func(void*)+200) (BuildId: <removed>)
```

Peter-St, 2023-12-30:
Hi ben,

I'm currently updating the automatic method of the app and made a mistake in the code. I just fixed it and you can give em another try.

Because of the pull, there shouldn't be any errors so far. If there is an error again, can u plz attach any log to this. Thx

ben-dempsey, 2023-12-30:
Hi Peter,

Thank you for the quick response!

I am still getting an error when I try to pull the latest commits, here is the error (using SourceTree git client):
`error: invalid path 'fastlane/metadata/android/de-DE/images/featureGraphic.'` <-- missing "jpeg" in file name

Peter-St, 2023-12-30:
Hi and thx for the hint.
Indeed there was the jpeg missing.
Give em another try 🙂

ben-dempsey, 2023-12-30:
I was able to pull the latest commit, and the crash has been fixed!

I do notice however that the only available frame rate option is 15 fps, when really my UVC camera supports 30 fps.
When I switch to use the "USB Alternative Driver" and go through the same flow I am able to see 30 fps as expected.

I tried the automatic method as well and it doesn't work for me, here is my log from after clicking _AUTOMATIC_ :
```
setSurface called with nullptr
setSurface() destroyed EGLSurface
destroyEglSurface
[ViewRootImpl@5c51253[SetUpTheUsbDeviceUvc]#3(BLAST Consumer)3](id:1fd300000003,api:1,p:8147,c:8147) disconnect: api 1
dispatchDetachedFromWindow
[ViewRootImpl@5c51253[SetUpTheUsbDeviceUvc]#3](f:0,a:1) destructor()
[ViewRootImpl@5c51253[SetUpTheUsbDeviceUvc]#3(BLAST Consumer)3](id:1fd300000003,api:0,p:-1,c:8147) disconnect
Input channel destroyed: 'd8558f7', fd=143
set:488 set: Unset optional value from type SMPTE2086
set:488 set: Unset optional value from type CTA861_3
onStateChanged: InsetsState: {mDisplayFrame=Rect(0, 0 - 720, 1600), mDisplayCutout=DisplayCutout{insets=Rect(0, 45 - 0, 0) waterfall=Insets{left=0, top=0, right=0, bottom=0} boundingRect={Bounds=[Rect(0, 0 - 0, 0), Rect(272, 0 - 448, 45), Rect(0, 0 - 0, 0), Rect(0, 0 - 0, 0)]} cutoutPathParserInfo={CutoutPathParserInfo{displayWidth=720 displayHeight=1600 physicalDisplayWidth=720 physicalDisplayHeight=1600 density={1.875} cutoutSpec={M 0,0 H -88 V 45 H 88 V 0 H 0 Z} rotation={0} scale={1.0} physicalPixelDisplaySizeRatio={1.0}}}}, mRoundedCorners=RoundedCorners{[RoundedCorner{position=TopLeft, radius=60, center=Point(60, 60)}, RoundedCorner{position=TopRight, radius=60, center=Point(660, 60)}, RoundedCorner{position=BottomRight, radius=60, center=Point(660, 1540)}, RoundedCorner{position=BottomLeft, radius=60, center=Point(60, 1540)}]}  mRoundedCornerFrame=Rect(0, 0 - 720, 1600), mPrivacyIndicatorBounds=PrivacyIndicatorBounds {static bounds=Rect(637, 0 - 720, 45) rotation=0}, mSources= { InsetsSource: {mType=ITYPE_STATUS_BAR, mFrame=[0,0][720,45], mVisible=true, mInsetsRoundedCornerFrame=false}, InsetsSource: {mType=ITYPE_NAVIGATION_BAR, mFrame=[0,1510][720,1600], mVisible=true, mInsetsRoundedCornerFrame=false}, InsetsSource: {mType=ITYPE_LEFT_GESTURES, mFrame=[0,0][0,1600], mVisible=true, mInsetsRoundedCornerFrame=false}, InsetsSource: {mType=ITYPE_RIGHT_GESTURES, mFrame=[720,0][720,1600], mVisible=true, mInsetsRoundedCornerFrame=false}, InsetsSource: {mType=ITYPE_TOP_MANDATORY_GESTURES, mFrame=[0,0][720,68], mVisible=true, mInsetsRoundedCornerFrame=false}, InsetsSource: {mType=ITYPE_BOTTOM_MANDATORY_GESTURES, mFrame=[0,1510][720,1600], mVisible=true, mInsetsRoundedCornerFrame=false}, InsetsSource: {mType=ITYPE_LEFT_DISPLAY_CUTOUT, mFrame=[0,0][-100000,1600], mVisible=true, mInsetsRoundedCornerFrame=false}, InsetsSource: {mType=ITYPE_TOP_DISPLAY_CUTOUT, mFrame=[0,0][720,45], mVisible=true, mInsetsRoundedCornerFrame=false}, InsetsSource: {mType=ITYPE_RIGHT_DISPLAY_CUTOUT, mFrame=[100000,0][720,1600], mVisible=true, mInsetsRoundedCornerFrame=false}, InsetsSource: {mType=ITYPE_BOTTOM_DISPLAY_CUTOUT, mFrame=[0,100000][720,1600], mVisible=true, mInsetsRoundedCornerFrame=false}, InsetsSource: {mType=ITYPE_TOP_TAPPABLE_ELEMENT, mFrame=[0,0][720,45], mVisible=true, mInsetsRoundedCornerFrame=false}, InsetsSource: {mType=ITYPE_BOTTOM_TAPPABLE_ELEMENT, mFrame=[0,1510][720,1600], mVisible=true, mInsetsRoundedCornerFrame=false} } host=humer.uvc_camera/humer.UvcCamera.SetUpTheUsbDeviceUvc from=android.view.ViewRootImpl$ViewRootHandler.handleMessageImpl:6740
begin
end
start preview automatic
begin
end (0)
begin
startPreview
test = -1
check test
preview_thread Test Mode
startPreview finished
end (0)
begin
format->guidFormat = -568153634
uvc_frame_format cf = 3
guid does not match
claiming interface 1
Interface sucessfully claimed
uvc_query_stream_ctrl
before the CTL
Control Transfer Sucessful
uvc_query_stream_ctrl
before the CTL
Control Transfer Sucessful
uvc_query_stream_ctrl
before the CTL
Control Transfer Sucessful
uvc_query_stream_ctrl
before the CTL
set:488 set: Unset optional value from type SMPTE2086
Control Transfer Sucessful
uvc_query_stream_ctrl
set:488 set: Unset optional value from type CTA861_3
before the CTL
Control Transfer Sucessful
uvc_query_stream_ctrl
before the CTL
Control Transfer Sucessful
_prepare_stream_ctrl sucessfully
_uvc_get_stream_ctrl_format failed
_uvc_get_stream_ctrl_format failed: -51
could not negotiate with camera:err=-51
end (-51)
The FrameFormat is == == == (null)
ctrl.bFormatIndex = 1
ctrl.bInterfaceNumber = 1
ctrl.dwFrameInterval = 666666
ctrl.bFrameIndex = 1
end
The imageWidth is 0
The FrameFormat is (null)
camera_pointer->imageWidth = 0
camera_pointer->frameFormat = (null)
preview automatic complete
The FrameFormat is (null)
camera.activeUrbs = 0
camera.packetsPerRequest = 0
camera.maxPacketSize = 0
camera.imageWidth = 0
camera.imageHeight = 0
camera.camFrameInterval = 666666
camera.frameFormat = null

```

Peter-St, 2023-12-30:
ok, I don't know, why the program not detects the 30 fps. In your log a `camFrameInterval = 666666` is detectet, which is equal to 15 FPS. I think it's a camera specific issue. Best solution will be to set it manual to 333333 with the edit Button and save these settings. Plz report if it works then :)

Second thing is the auto method. The method is in beta for now and will be getting better in Time. I noticed, that the method is close to work even with your cam, although there is an error in `_uvc_get_stream_ctrl_format` in the LibUvc library. I will try to fix this, but it will take some time.

ben-dempsey, 2024-01-02:
When I go through the manual settings and then edit/save the Frame_Interval as 333333, it appears to be correct for the control transfer but then there are still only ~70 frames captured in 5 seconds which is roughly close to 15fps.

Screenshots:

[image]

[image]