Peter-St/Android-UVC-Camera #42: Saving image with error

MustafaErkinDal, 2024-02-14:
Hi,

I am using Pure Thermal 2 Flir 3.5. I have no problems with camera stream. However, when I save an image it saves in wrong colors (purple and green) with stripes on it. I am using Android Studio. Have you ever face such a problem ? Am I doing something wrong ?

Thanks in advance.  

[image]

My settings are:

Packets Per Request = 4
Active Urbs = 4
AltSetting = 1
Maximal Packet Size = 642
Videoformat = YUv2
Camera Format Index = 1
Camera Framelndex = 1
Image Width = 160
Image Height = 120
Camera Frame Interval (fps) = 9
UVC Values:
bUnitID = 0
bTerminaliD = 0
bcdUVC = 1.00
LibUsb = true

Peter-St, 2024-02-14:
Hoi Mustafa,

I noticed that in the method 'PreviewPrepareStream' in the 'uvc_support.cpp' file there is the part for YUV video format missing (in line 98 - 102). I think that was because YUV is not that often much in use. I will fix it in the next couple of days (depends on the time I have) and then come back to you to test it.

Wish you a happy carnival,

Peter

Peter-St, 2024-02-18:
You need to change your Videoformat to UYVY.
Do so and report if it worked.

MustafaErkinDal, 2024-02-21:
Hello Peter,

I changed the format to UYVY as you said but sadly it did not work. If you may look at the StartIsoStreamActivityUvc.java file line 1302 it is like the following screenshot

[image]

I assume UYVY format is purposely out of use. I tried adding "PreviewPrepareStream(mNativePtr, mPreviewSurface, new IFrameCallback()..." method, copying from other formats code, but I could not see the camera stream. If you have any other suggestions I will be appreciated.

Peter-St, 2024-02-21:
Hoi,

Your hints were right and I made some changes to solve the issue.
Can you plz update your repo and test out the stream again.
Thx, Peter