Peter-St/Android-UVC-Camera #43: V2.2.4 Video recording does not seem to be working

EPadronU, 2024-02-22:
## Steps to reproduce - Workflow 1
1. Tap on `TRANSMISSION START`
2. Tap on `Start the camera`
3. Tap on the `REC` red button (it changes to a `Stop icon` red button)
4. Tab on the `Stop icon` red button  (it changes back to the `REC` red button)

### Expected
1. A message saying the video was successfully captured
2. Such a video being present somewhere in the phone's internal memory

### Actual
Nothing happens

## Steps to reproduce - Workflow 2
1. Tap on `TRANSMISSION START`
2. Tap on `Start the camera`
3. Long tap on the `REC` red button (it changes to a `Stop icon` red button)
4. Two messages appears as popups one after the other: `Long Click - Video starts` and `Record started`
5. Tab on the `Stop icon` red button (it changes back to the `REC` red button)

### Expected
1. A message saying the video was successfully captured
2. Such a video being present somewhere in the phone's internal memory

### Actual
Nothing happens

## Additional information

Picture/photo capture works as expected

Peter-St, 2024-02-25:
HI,

I fixed the video record.
Can you plz try it out and report if it works.
Sadly with my cam I get a darker Video, when I record the stream. The record is performed over the BitmapToVideoEncoder.java file. -> There in Line 360 the Video Format is chosen. I tried out some different, but non of them worked so far. Perhaps you find a solution with a different COLOR_Format ;)

So far,

Peter

EPadronU, 2024-02-26:
Hi Peter,

Could you provide to me the APK file? I'd like to avoid installing everything that's required for doing it myself, since it's a lot and I won't need it for anything else.

Peter-St, 2024-02-28:
Hoi,

i will try to fix the brightness error the next couple of days and will then upload the app to F-Droid and PlayStore.

MakeASubfolderDownloadLink, 2024-06-20:
Really looking forward to trying the new apk. I'm out of ideas to make video recording work. Thanks for all the work you have done on this, it's making a huge difference where it's needed!

Peter-St, 2024-06-24:
Hi,
you can try this apk --> [app-release.apk.zip](https://github.com/user-attachments/files/15960290/app-release.apk.zip) (delete the .zip from the attached file and install the apk). The video record should work, but there are issues with the brightness. If you can life with it for now, you can use this apk. Perhaps I can fix this issue in the future.

So far,

Peter